steveicarus/iverilog #854: Failure to build with non ansi port declarations

antonblanchard, 2023-01-05:
In https://github.com/efabless/caravel/issues/402 an update of Icarus Verilog resulted in a compile error. A cut down test case:

```
module test (
  input i,
  output o
);

  wire i;
  reg o;

endmodule
```

The error reported by iverilog:

```
test.v:6: error: 'i' has already been declared in this scope.
test.v:2:      : It was declared here as a net.
test.v:7: error: 'o' has already been declared in this scope.
test.v:3:      : It was declared here as a net.
```

This was bisected to https://github.com/steveicarus/iverilog/commit/6204b78610fedbe36b73fd26bf144f2f07849fb3

Assuming this syntax is "non-ANSI" (I don't deal with a lot of Verilog so I wasn't 100% sure), then it looks like an issue with the parser where an implicit net type ends up marking the port as `SR_BOTH`.

larsclausen, 2023-01-05:
LRM explicitly declares the code above as not valid

> 23.2.2.2 ANSI style list of port declarations
> 
> Each port declaration provides the complete information about the port. The port’s direction, width, net or
> variable type, and signedness are completely described. The port identifier **shall not** be redeclared, in part or
> in full, inside the module body.

ANSI style declaration is when your port list is in parenthesis. Non-ansi when it is as separate statements.
```SystemVerilog
// ANSI
module test(input i, ouput o);
...
endmodule

// Non-ANSI
module test;
input i;
output o;
...
endmodule
```
For the latter case it is allowed to also have a `wire i` in there, for the former not.

I'm a bit torn here. On one hand this is breaking existing code, so it might make sense to have a compiler option to allow this. On the other hand it is clearly not allowed by the standard and not just a gray area. And other tools also will report an error.

antonblanchard, 2023-01-05:
@larsclausen Thanks for the clarification. I just checked Verilator and it fails too, so it seems like the caravel verilog should be fixed.

antonblanchard, 2023-01-05:
I created a PR to fix the caravel issues: https://github.com/efabless/caravel/pull/403

martinwhitaker, 2023-04-16:
As caravel has been fixed, closing as invalid. We can reconsider if we get a lot more complaints.

closed 2023-04-16 by martinwhitaker