steveicarus/iverilog #867: -y option is not working

catkira, 2023-01-29:
The -y option is very strangely not working for me, its probably a stupid mistake on my side, but I cant find it
```
iverilog -o /mnt/d/git/intfftk/sim_build/NFFT=7/int_fft_ifft_pair.vvp -D COCOTB_SIM=1 -g2012 -s int_fft_ifft_pair -Pint_fft_ifft_pair.NFFT=7 -s glbl -Y .v -y /mnt/d/git/intfftk/src/verilog/XilinxUnisimLibrary /mnt/d/git/intfftk/src/verilog/main/int_fft_ifft_pair.v /mnt/d/git/intfftk/src/verilog/fft/int_dif2_fly.v /mnt/d/git/intfftk/src/verilog/fft/int_dit2_fly.v /mnt/d/git/intfftk/src/verilog/fft/int_fftNk.v /mnt/d/git/intfftk/src/verilog/fft/int_ifftNk.v /mnt/d/git/intfftk/src/verilog/math/int_addsub_dsp48.v /mnt/d/git/intfftk/src/verilog/math/cmult/int_cmult_dsp48.v /mnt/d/git/intfftk/src/verilog/math/cmult/int_cmult18x25_dsp48.v
```
gives me this error
```
error: Unable to find the root module "glbl" in the Verilog source.
     : Perhaps ``-s glbl'' is incorrect?
```
Does anybody have an idea why it is not working? glbl.v is definately present 

```
benja@DESKTOP:/mnt/d/git/intfftk$ ls /mnt/d/git/intfftk/src/verilog/XilinxUnisimLibrary
glbl.v  unisims
benja@DESKTOP:/mnt/d/git/intfftk$
```

catkira, 2023-01-29:
It works when I use `-sglbl` instead of `-s glbl` and specify the source files explicitely without using `-y`
```
iverilog -o /mnt/d/git/intfftk/sim_build/NFFT=7/int_fft_ifft_pair.vvp -D COCOTB_SIM=1 -g2012 -s int_fft_ifft_pair -Pint_fft_ifft_pair.NFFT=7 -sglbl /mnt/d/git/intfftk/src/verilog/main/int_fft_ifft_pair.v /mnt/d/git/intfftk/src/verilog/fft/int_dif2_fly.v /mnt/d/git/intfftk/src/verilog/fft/int_dit2_fly.v /mnt/d/git/intfftk/src/verilog/fft/int_fftNk.v /mnt/d/git/intfftk/src/verilog/fft/int_ifftNk.v /mnt/d/git/intfftk/src/verilog/math/int_addsub_dsp48.v /mnt/d/git/intfftk/src/verilog/math/cmult/int_cmult_dsp48.v /mnt/d/git/intfftk/src/verilog/math/cmult/int_cmult18x25_dsp48.v /mnt/d/git/intfftk/src/verilog/XilinxUnisimLibrary/glbl.v /mnt/d/git/intfftk/src/verilog/XilinxUnisimLibrary/unisims/DSP48E1.v /mnt/d/git/intfftk/src/verilog/XilinxUnisimLibrary/unisims/DSP48E2.v
```
this works, but this doesn't
```
iverilog -o /mnt/d/git/intfftk/sim_build/NFFT=7/int_fft_ifft_pair.vvp -D COCOTB_SIM=1 -g2012 -s int_fft_ifft_pair -Pint_fft_ifft_pair.NFFT=7 -s glbl -y /mnt/d/git/intfftk/src/verilog/XilinxUnisimLibrary -y /mnt/d/git/intfftk/src/verilog/XilinxUnisimLibrary/unisims /mnt/d/git/intfftk/src/verilog/main/int_fft_ifft_pair.v /mnt/d/git/intfftk/src/verilog/fft/int_dif2_fly.v /mnt/d/git/intfftk/src/verilog/fft/int_dit2_fly.v /mnt/d/git/intfftk/src/verilog/fft/int_fftNk.v /mnt/d/git/intfftk/src/verilog/fft/int_ifftNk.v /mnt/d/git/intfftk/src/verilog/math/int_addsub_dsp48.v /mnt/d/git/intfftk/src/verilog/math/cmult/int_cmult_dsp48.v /mnt/d/git/intfftk/src/verilog/math/cmult/int_cmult18x25_dsp48.v /mnt/d/git/intfftk/src/verilog/XilinxUnisimLibrary/glbl.v /mnt/d/git/intfftk/src/verilog/XilinxUnisimLibrary/unisims/DSP48E1.v /mnt/d/git/intfftk/src/verilog/XilinxUnisimLibrary/unisims/DSP48E2.v
```
Can somebody explain this?

martinwhitaker, 2023-01-29:
-y specifies a search path for library modules, i.e. modules that are instantiated by some other module. You have to specify the source files for top-level modules explicitly. -s merely allows you to specify which of the available top-level modules are elaborated, which can be useful if some of your source files contain more than one module and you don't use (and don't want to elaborate) all of those modules.

catkira, 2023-01-29:
ok I understand, thank You!

closed 2023-01-29 by catkira