steveicarus/iverilog #871: Compilation failure due to #delay and Usigned in the always block

cemery123, 2023-02-20:
Hello, we found a compilation failure problem during the running process. Just like the example we submitted (sorry that my special reduction tool can only manually reduce to the simplest situation we can reduce), we found that delay and Unsigned in one expression （in line 287）will cause compilation errors. It seems that this approach does not violate the grammar of Verilog, we want to find out the existence of these problems
Our files is in https://drive.google.com/drive/folders/17H_IMI2QZZeg1---rxyj4jG0jzryKzb2?usp=sharing 
and https://drive.google.com/file/d/1T-ny2dcKUucz39IHCk2usViblILajsoD/view?usp=sharing

larsclausen, 2023-02-25:
Can you share the error message you are seeing?

cemery123, 2023-02-25:
sure, you can repeat by the file i have shared.Iverilog has an error like follow:
ivl: net_link.cc:130: Nexus* Link::find_nexus_() const: Assertion 'next_' failed.
I think not only singed but unsight looks like arsing the problem.So,is that unsported by Iverilog?

cemery123, 2023-02-25:
sure, you can repeat by the file i have shared.Iverilog has an error like follow:
ivl: net_link.cc:130: Nexus* Link::find_nexus_() const: Assertion 'next_' failed.
I think not only singed but unsight looks like arsing the problem.So,is that unspported by Iverilog?

cemery123, 2023-02-25:
sure, you can repeat by the file i have shared.Iverilog has an error like follow:
ivl: net_link.cc:130: Nexus* Link::find_nexus_() const: Assertion 'next_' failed.
I think not only singed but unsighed looks like arsing the problem.So,is that unspported by Iverilog?

larsclausen, 2023-02-26:
This looks like a duplicate of #793 which you create a while ago. A fix was merged in #795. Try to use the latest development version the issue should be fixed in there.

martinwhitaker, 2023-04-16:
Assuming fixed by #795.

closed 2023-04-16 by martinwhitaker