steveicarus/iverilog #876: Possibilities miss simulation

cemery123, 2023-03-03:
Hello, we found a puzzling problem in the actual operation of the project. In order to ensure the simplicity of the code, we have merged the part of the inertial delay. In theory, this approach will not affect the output of the result. Inertial delay will not be detected due to simulation characteristics. However, during our actual operation, we found that Iverilog had data exceptions in some files after the merger. It is abnormal if we don't add time period limit to pass value, but if we add time period limit eg` `timescale 1ns/1ps`The result is the same. This situation confuses us. As shown in the document we submitted, Iverilog may have an exception in the value transfer of the unsigned module of the underlying module, resulting in a slight change in the overall simulation output.
You can reproduce this problem by running our files, where the file with _mts suffix is our original stimulus file, and the one without this is our merged file. Rtl.v is the design file
Our files were in follows:
https://drive.google.com/drive/folders/1HTokRcpYVVytGJqypbwVy-lHAIO6TGVL?usp=sharing

cemery123, 2023-03-06:
https://drive.google.com/drive/folders/1HTokRcpYVVytGJqypbwVy-lHAIO6TGVL?usp=sharing

martinwhitaker, 2023-04-16:
I'm assuming this is the same fundamental issue as #884. Feel free to reopen if you can provide a **simple** test case that reproduces the problem.

closed 2023-04-16 by martinwhitaker