steveicarus/iverilog #878: Multiple systemverilog files

armstrong0, 2023-03-05:
I have a set of four systemverilog files,  two are types files and two contain systemverilog logic that I will eventually synthesize.  The file list is **sd_types,.svh sdspi_types.svh, sdspi.sv and sd.sv.**  The top module is sd.sv.

The two sv files contain **import** statements but no includes.

with verilator the following command works (parses correctly and generates c++.) 
    **verilator -cc  --top-module sd  sdspi_types.svh sd_types.svh sdspi.sv sd.sv** 
similarly verible-verilog works correctly ( with lots of lint warnings, but no errors)
     **verible-verilog-lint sdspi_types.svh sd_types.svh sd.sv**

I am able to successfully convert these systemverilog files to verilog with: 
     **sv2v    sd_types.svh sdspi_types.svh sdspi.sv sd.sv > sd.v**

I would like to be able to simulate this logic with iverilog but have been unable to get it to accept the list of files without a syntax error! I expect that it is simply me misunderstanding how to do this...  of course there would be a test bed file as well but until iverilog accepts the file list there is no point...

iverilog -g2012 ....

larsclausen, 2023-03-05:
To better understand the problem can you share those files, or a simplified version of them that still reproduces the issue?

In general multiple files are support. There might be a specific construct in your files that is triggering the problem.

armstrong0, 2023-03-05:
It is non-trivial to strip the files down.  I will include the full version.  
[full_set.tar.gz](https://github.com/steveicarus/iverilog/files/10892926/full_set.tar.gz)

larsclausen, 2023-03-05:
I think that might be just a ordering issue. iverilog requests that package is declared before it is reference. Otherwise it generates 'syntax error'. Which of course is a bad error message. It should instead say that it can't find the package.

Running iverilog with your files like `iverilog -g2012 sdspi_types.svh sd_types.svh sdspi.sv sd.sv` seems to work. This is with the latest github build. If you are using an older version it might also be that imports in the module header are not supported in that version.

armstrong0, 2023-03-05:
OK I am running V11 (stable)
The error I get using the command line you used is:

iverilog -g2012 sdspi_types.svh sd_types.svh sdspi.sv sd.sv 
sd_types.svh:52: sorry: I don't know how to elaborate 13vector_type_t here.

I will update to the latest github build.   

Thank you very much for the prompt response!

martinwhitaker, 2023-04-16:
I'm assuming updating to the latest code has resolved this issue, so closing.

closed 2023-04-16 by martinwhitaker