steveicarus/iverilog #883: vcd2fsdb failed

yangguangguang, 2023-03-13:
Hi:

  I`m running the [verilog-axi](https://github.com/alexforencich/verilog-axi) testbenches and use the following configuration to generate the fst file：

```
ifeq ($(WAVES), 1)
VERILOG_SOURCES += iverilog_dump.v
COMPILE_ARGS += -s iverilog_dump
endif

iverilog_dump.v:
	echo 'module iverilog_dump();' > $@
	echo 'initial begin' >> $@
	echo '    $$dumpfile("$(TOPLEVEL).fst");' >> $@
	echo '    $$dumpvars(0, $(TOPLEVEL));' >> $@
	echo 'end' >> $@
	echo 'endmodule' >> $@
  
```

and then I use 'fst2vcd' command converts fst file to vcd, I can use GTKWave to view all waveforms. But when I use 'vcd2fsdb' command converts vcd file to fsdb, and use Verdi to view waveforms, then I can only view the  three layers waveforms of the fsdb file.The warning of the Verdi is that failed to add certain signals to waveform because they cannot be found in FSDB.

what could be the cause of the problem？Thanks

martinwhitaker, 2023-03-13:
Do you get different results if you let vvp generate the VCD file itself?

yangguangguang, 2023-03-14:
I tried a new method and it worked：
1: use the 'fsdbdebug' tools to view the *.fsdb to ensure that the waveform of the signal is present.
2: open verdi like this:
[image]
I found that a '\\' prefix is ​​added in front of the m_ifaces[0], but in the Hierarchy Browser Pane, I can't see the '\\' prefix added in front of m_ifaces[0]. in the *.fsdb, I found that m_ifaces[0] is recorded like this:
[image]
Whenever I add a signal waveform through the Hierarchy Browser Pane, verdi prompts an error message("Failed to add certain signals to waveform because they cannot be found in FSDB"). When I checked the original vcd file, I found no abnormal phenomenon:
[image]
rtl uses 'generate for' to create instantiations of modules, not sure if this is the reason, thanks a lot!

martinwhitaker, 2023-04-16:
This doesn't appear to be a bug in Icarus, and the user has found a solution, so closing.

closed 2023-04-16 by martinwhitaker