steveicarus/iverilog #884: Misssimulation in reset process

cemery123, 2023-03-13:
Hello, we found a puzzling problem in the actual operation of the project. In order to ensure the simplicity of the code, we have merged the part of the inertial delay. In theory, this approach will not affect the output of the result. Inertial delay will not be detected due to simulation characteristics. However, during our actual operation, we found that Iverilog had data anomalies in some of the merged files. As shown in the document we submitted, Iverilog may have an exception in the value transfer of the reset process
You can reproduce this problem by running our files, where the file with _mts suffix is our original stimulus file, and the one without this is our merged file. Rtl.v is the design file
https://drive.google.com/drive/folders/1lkybRSGaOsaX4L-yVGdQytyFDtu8fwQi?usp=sharing

martinwhitaker, 2023-03-13:
I would need a much simpler test case before I started looking at this. Please try to reduce your code to the minimal amount needed to demonstrate the problem. Also, please attach your code directly here (rather than putting it on an external site), and describe both the expected result and the actual result.

cemery123, 2023-03-13:
Fine,even the simplest code have a big scale.I have try to reduce it but it does not work.I will put the minimal vision in fllows.for your convenient,I aslo advice you download the code from google drive because the code which in drive have original message.(We will also continue to work hard to simplify our questions, and we will let you know in the comments if there are new developments)As the issue i putted in last week,style of these problem is misssmuliation.I have attached reslut of the problem you can see it just have small incorrect in the reslut,as The part circled by the red wireframe.
![1678700381605](https://us
[miss.zip](https://github.com/steveicarus/iverilog/files/10955793/miss.zip)
er-images.githubusercontent.com/63329182/224664066-2a2afac2-ffe0-4319-9a7f-b5354384a073.png)

cemery123, 2023-03-15:
I have reduced the syn file to minimal.As you can see the miss simulation is still exists
[syn_identity.zip](https://github.com/steveicarus/iverilog/files/10975233/syn_identity.zip)

martinwhitaker, 2023-03-15:
I tested your code with another simulator. It gave the same results as Icarus. So I think you need to look at your code again.

cemery123, 2023-03-15:
Sorry, can you show me the results of your run. We verified the results through vivado. Unfortunately, he did not perform the same as iverilog. What's more, incorporating the inertial delay of skip detection should not change the result. Is such an operation not allowed?

martinwhitaker, 2023-03-18:
I have confirmed that the simulator provided with Vivado gives the same results for both cases. But using the simulator provided with another FPGA vendor's design tools gives the same result as Icarus. I would suspect there's a race in your code. You could try some more simulators by visiting https://edaplayground.com/

I have no idea what you mean by "incorporating the inertial delay of skip detection". Please try to write a simple and clear explanation of what you are trying to do and provide a properly short and non-obfuscated test case.

martinwhitaker, 2023-04-16:
No response to my request for better information, so closing.

closed 2023-04-16 by martinwhitaker