steveicarus/iverilog #885: Function input name "type" causes syntax error

geoskd, 2023-03-18:
The attached file has a simple demo showing the problem code.

I verified that this code compiles without errors using https://www.tutorialspoint.com/compile_verilog_online.php as well as the Efinity built-in verilog compiler.

I have a work-around for icarus by simply changing the input name from "type" to "typ", but this is less than ideal since the source code in question is the output of a third party verilog generator tool.

[test.v.txt](https://github.com/steveicarus/iverilog/files/11007215/test.v.txt)

larsclausen, 2023-03-18:
How do you run the `iverilog` command? `type` is a keyword in SystemVerilog. If you run iverilog in SystenVerilog mode the error is expected. In Verilog mode it should work.

```
# iverilog test.v
# iverilog -g2012 test.v
foo.v:7: syntax error
foo.v:5: error: Syntax error defining function.

```

geoskd, 2023-03-18:
> How do you run the `iverilog` command? `type` is a keyword in SystemVerilog. If you run iverilog in SystenVerilog mode the error is expected. In Verilog mode it should work.
> 
> ```
> # iverilog test.v
> # iverilog -g2012 test.v
> test.v:7: syntax error
> test.v:5: error: Syntax error defining function.
> ```

I am running with -g2009. Many of my 3rd party libraries require at least 2005-sv, and all of them claim to be compatible with systemverilog up to 2012. The Efinix' (also provider of the problem library) Efinity verilog synthesizer also claims to be compliant with the 2012 standard. Because of these requirements, I *must* compile with at least 2005-sv.

This is one of those ugly corner-cases that the standards folks missed. Technically, the compiler behavior in this case is not defined by the specification, and I have also put a request in to the library authors to rename the variable to something less fraught.

One final note: The "type" name in this particular case can't possibly be the systemverilog "type" keyword because an actual "integer" type has already been provided. As such, the only thing the "type" could possibly be in this context is the actual name of the input. Throwing a syntax error is technically acceptable according to the 2012 standard, but it is also acceptable under that standard to simply allow the use of "type" as the name of the input. Although both approaches match the standard, the former is a bit more user-friendly as it puts fewer constraints on the end-user.

Given that it is not at all obvious from the standard why the attached code fails to compile with Icarus, I would recommend one of the following two remedies for this issue:
1. allow the use of the "type" as an input name in this specific context to match the other standards compliant implementations. This also matches the spirit that sv is supposed to be backwards compatible with older verilog standards.
2. Improve the error message for this instance to make clear that the cause of the syntax error is the "type" being interpreted as a keyword and being consequently disallowed as an input name. This is less clean than option 1, but should be simpler to implement, and will at least give people something to work with when their code compiles on other SV compilers, but not Icarus.

larsclausen, 2023-03-18:
I agree, the error message can and should be better.

But the standard is pretty clear on that keywords are reserved and can not be used as identifier names. Same way `integer if;` or `integer integer;` are invalid syntax `integer type;` is invalid syntax. Some tools might allow this, for example because they do not implement the `type` keyword.

The standards authors did consider this case and they to provide a mechanism to disable certain keywords by using the `` `begin_keywords`` directive. It allows to specify on a per file basis keywords from which version should be recognized. You can for example place `` `begin_keywords "1364-2005"`` at the beginning of your file and `` `end_keywords`` at the end to disable SystemVerilog keywords and only recognize Verilog keywords. In this case you can use `type` as an identifier name.

geoskd, 2023-03-18:
My point was that you don't actually need to disable the "type" keyword in order for the code to be 100% backward compatible because you can deduce from the context whether the use of "type" is intended to be a systemverilog keyword or a name. If the user provided an existing type specification according to the original verilog standard, then any following text must be a name regardless of whether it matches existing keywords or not. I realize that making the compiler understand this particular bit of context is not trivial, so I agree with the decision to keep it as a syntax error with improved error messaging.

A similar problem occurred with C++ superseding the C programming language and all of the new keywords that C++ introduced. In order to maintain 100% backward compatibility, C++ compilers had to move to a 100% context aware keyword processing so that they could tell the difference between keywords or user defined names. This change necessitated a pretty significant change to the language parser, but it was the only way for C++ compilers to handle the incredible variety of shady-but-still-standards-compliant C code that exists to this day. Similarly, there is a lot of legacy verilog library code that uses "type" as a variable name. Being able to use that legacy code in combination with more modern code is and was a big part of the goals of systemverilog.

Anyway, even with that, I agree that option 2 is the more straight-forward option here. We can agree to disagree about whether option 1 accurately represents the standard or not, but by selecting option 2, the question is moot.

martinwhitaker, 2023-04-16:
I agree with @larsclausen , the (System)Verilog mechanism for supporting backwards-compatibility is and always has been the `` `begin_keywords `` directive.

With the current flex/bison based parser, it would be very hard to support context-aware keyword processing. Even providing better error messages for cases like this is a lot of work.