steveicarus/iverilog #887: The result is abnormal caused by adding delay before Unsigned

cemery123, 2023-04-03:
Hello, we found a puzzling problem during the actual operation of the project. In order to ensure the conciseness of the code, we merged the part of the inertial delay. Theoretically, this approach will not affect the output of the results. Because the simulation characteristics cause the inertial delay to not be detected. However, during our actual operation, we found that data anomalies appeared in some files after the merger. In order to facilitate your reproduction of the problem, we successfully reduced the 1200-line project code to 200 lines. As shown in the file we submitted, if you cancel the delay of the last line of code, Iverilog will output consistent values. Adding a delay before unsigned will cause the entire output result to be unpredictable. The file with the suffix _mts is our original stimulus file, and the one without this is our merged file. Rtl.v is a design file.
[image]

[image]

[work.zip](https://github.com/steveicarus/iverilog/files/11135752/work.zip)

red0bear, 2023-04-03:
hello there , friend are you trying run this on fpga ? 

1.Actual code sound no friendly , can you explain what he does ? Did you auto generated that ?
2.Can you fix your includes ?

martinwhitaker, 2023-04-16:
I can't reproduce the issue. As this looks very similar to #884, I suspect this is a race in your code. So unless you provide a **much** simpler test case that demonstrates the problem, I'm going to assume that is the case.

closed 2023-04-16 by martinwhitaker