steveicarus/iverilog #889: SDF warning/error enhancement proposal

svenka3, 2023-04-12:
I understand SDF support is limited and there is a GSOC project open as well. I came across a (very) low-hanging fruit type of enhancement and I have a local patch. Want to know if you are interested in a PR.

In my case, the DUT instance to be annotated is specified only inside SDF file, not during $sdf_annotate call (in Verilog). Wo when the instance name (or module name) is misspelt inside SDF I get:

```log
SDF WARNING: ../vlog_delay_mode_sdf.sv:42: Unable to find inst1 in scope dut_inv_with_delays. 
```
That line 42 in Verilog looks as follows:
```verilog
    $sdf_annotate("sample.sdf");  //, inst);
```
Here vvp found an instance named inst1 and that source is actually SDF, relevant line below:

```verilog
  (INSTANCE inst1)
```
So, a first-timer would be lost as to where the fix shall be. I enhanced it to pass SDF line number from sdf_parse.y to sys_sdf.c and have the below:

```log
SDF WARNING: ../vlog_delay_mode_sdf.sv:42: Unable to find inst1 in scope dut_inv_with_delays. SDF: sample.sdf 5
```
It is no rocket science, but does improve the usability IMHO. Also it is not perfect as it prints the line-number of CELL than the INSTANCE, but nevertheless it is showing the SDF file name and a close line. 

I do intend to enhance few more such low-hanging stuff around this area if this is of interest.

Thanks

red0bear, 2023-04-12:
System verilog is limited too , did you try verilog first ?

svenka3, 2023-04-12:
> System verilog is limited too , did you try verilog first ?

Well, SV is not that relevant in this thread - apart from my file indeed using endlabels etc. But, yes even in pure Verilog the same issue/enhancement applies.

Thanks

red0bear, 2023-04-12:
i remember i try use SDF past to test long time ago and it have things missing. You can make some functional verification using verilog/PLI/VPI .

martinwhitaker, 2023-04-16:
@svenka3 , it looks like a useful enhancement to me.

svenka3, 2023-04-16:
Thanks. I now have a local branch with this tiny enhancement (after git pull/fetch/sync). Reading this:
https://github.com/steveicarus/iverilog/blob/master/Documentation/developer/getting_started.rst

Should I run regression and/or update some GOLD files to ensure this enhancement does not fail unintentionally? Sorry to ask as it is not exactly clear to me how this ivl-regression system is made self-checking.

Thanks

svenka3, 2023-04-16:
Also, on regression itself - I recall you moved/moving to PY based. The instructions say:

% cd ivtest
% ./vvp_reg.pl --strict

Should I run using PL or PY please?

svenka3, 2023-04-18:
Thanks. I will add extra tests as I find it necessary soon. One observation on the regression, specifically on "perl vvp_reg.pl"

I believe it uses "iverilog" as-is, meaning if I have a local build of latest-and-greatest iverilog, this script may not pick-it-up, instead it picks from from my $PATH. Is my understanding correct? If so, few requests/suggestions:

1. Print which iverilog is being used in "regression_report.txt" and/or to the screen.
2. Enhance it to optionally choose which one (though I can prefix $PATH etc.). 

Comments please?

caryr, 2023-06-26:
I have pushed changes that add the SDF file/line information to the various messages. I'm thinking about hte best way to deal with the iverilog/vvp path. I make make this part of -V