steveicarus/iverilog #892: Trying to run PLI program, "thunker register" issues

r2com, 2023-04-15:
So I have a basic verilog module and I compiled without issues the verilog PLI C progarm which registers callbacks and scans for ports. both verilog rtl and C compile OK, however when I run it I see:

```
$ vvp -mcadpli dut_top -cadpli=../vip/build/libvpi.vpi.so                                                                                                                                                                                                                                                                                            
vvp: cadpli.c:50: thunker_register: Assertion `bp' failed.
Aborted    
```

it doesnt say much about where should I look for, any hints?

red0bear, 2023-04-15:
hello, i dont like to auto promote me here ; i have some examples on https://github.com/GLADICOS i have been use. i suggest you look if you can a book called [Stuart_Sutherland] The Verilog PLI Handbook

r2com, 2023-04-15:
> hello, i dont like to auto promote me here ; i have some examples on https://github.com/GLADICOS i have been use. i suggest you look if you can a book called [Stuart_Sutherland] The Verilog PLI Handbook

I am sorry but your response is absolutely useless and has nothing to do with my original question with detailed description of problem. I _do_ have that book in fact. And I did not post a question "how do i VPI to Verilog" or anything like that.

red0bear, 2023-04-15:
useless is me wasting time with rude people. But anyway as you make this kind o response i see you put too little effort to resolve your problems and want quick solve by time of others. be less lazy person and start read more before come here search quick solution.

martinwhitaker, 2023-04-15:
> it doesnt say much about where should I look for, any hints?

From the assertion failed message, we need to look at line 50 in cadpli.c. There I see the code is parsing the `-cadpli=` argument and is failing because it doesn't contain a colon. Yes, this should be an error message, not an assertion, but you are using the old PLI interface that was removed from the Verilog language ~20 years ago, so I guess this hasn't had much attention.

The correct usage of the -cadpli= argument is described in the "Cadence PLI Modules" section of https://steveicarus.github.io/iverilog/usage/vpi.html

r2com, 2023-04-15:
> but you are using the old PLI interface that was removed from the Verilog language ~20 years ago

At the beginning I did not get why you thought I'm using 20 year old library, but I think it's because you noticed me doing "cadpli" option in a command line. However, honestly that was not my intent, I am not using TF or ACC routines I am using PLI 2.0 VPI routines, so I was not even supposed to use that command option.

What actually now worked for me is doing this:
`vvp -M ../vip/build/ -m dut_top ./dut_top.vvp   `
Now it all works for me.

However, one annoying thing here is that whatever passed after "-m" option is being added a ".vpi" extension, so in my case the C program had to be compiled to "dut_top.vpi" filename for it to work. 

I wonder if there is a way for iverilog not to append internally extensions to the files im pointing in command line, but in any case its a minor thing, the main issue is solved.

martinwhitaker, 2023-04-15:
Yes, the cadpli module only supports PLI1, and I assumed you had chosen to use it for that reason.

As mentioned in the `vvp` man page, if you specify a full path in the `-m` argument , the .vpi suffix won't be automatically appended.

closed 2023-04-16 by r2com