steveicarus/iverilog #897: SVA: immediate assert throws syntax error

svenka3, 2023-04-16:
I am trying a simple, immediate SVA in Icarus (main/devel). I get syntax error on the code below. Is immediate assert easy enough to add support for? Also how do I enhance the error if I wish to?

Thanks

```log
t.sv:7: syntax error
t.sv:7: error: Malformed statement
```

```verilog
module m;

  reg [7:0] a;

  initial begin
    #100;
    a1 : assert (!$isunknown(a));
    #10;
    $finish(2);
  end
endmodule : m
```

svenka3, 2023-04-16:
Tried few variants - and realized that label before the assert is the root cause. Should be easy to fix in parse.y - am I looking at right place?

martinwhitaker, 2023-04-17:
Yes, parse.y is the place to fix this.

svenka3, 2023-04-18:
Thanks, I have a fix as below:

```yacc
procedural_assertion_statement /* IEEE1800-2012 A.6.10 */
  : block_identifier_opt concurrent_assertion_statement
      { $$ = $2; }
  | block_identifier_opt simple_immediate_assertion_statement
      { $$ = $2; }
  | block_identifier_opt deferred_immediate_assertion_statement
      { $$ = $2; }
  ;
```

Test:
```verilog
module m;

  reg [7:0] a;

  initial begin
    #100;
    assert (!$isunknown(a));
    a1 : assert (!$isunknown(a)) else $error ("a1");
    #10;
    $finish(2);
  end
endmodule : m
```

It seems to work fine, should I raise a PR?

Log:

```log
ERROR: t.sv:7:
       Time: 100  Scope: m
ERROR: t.sv:8: a1
       Time: 100  Scope: m
t.sv:10: $finish(2) called at 110 (1s)
```

PR #906 (2023-06-11): Added support for labels in immediate assert/assume
Updated parse.y to add labels. Also added labelled assert/assume in existing tests. Updated GOLD files accordingly. Fixes #897
closed 2023-06-11 by caryr