steveicarus/iverilog #901: configure fails: error: /usr/local/bin/autoheader2.71 failed with exit status: 1

yurivict, 2023-04-17:
While trying to update the FreeBSD cad/icarus port to 12.0 there is this failure:
```
autoheader2.71: warning: missing template: HAVE_FMAX
autoheader2.71: warning: Use AC_DEFINE([HAVE_FMAX], [], [Description])
autoheader2.71: warning: missing template: HAVE_FMIN
autoheader2.71: warning: missing template: HAVE_LLROUND
autoheader2.71: warning: missing template: HAVE_LROUND
autoheader2.71: warning: missing template: HAVE_NAN
autoheader2.71: warning: missing template: UINT64_T_AND_ULONG_SAME
autoreconf2.71: error: /usr/local/bin/autoheader2.71 failed with exit status: 1
```

The GitHub tarball is used.

What is wrong?

Is the FTP site ftp://icarus.com/pub/eda/verilog/v11/ with tarballs gone?

autoconf-2.71
automake-1.16.5

FreeBSD 13.2

martinwhitaker, 2023-04-17:
> What is wrong?

What command are you running to get that error?

> Is the FTP site ftp://icarus.com/pub/eda/verilog/v11/ with tarballs gone?

See #886. I don't know if this is temporary or permanent. Only @steveicarus can answer that.

yurivict, 2023-04-17:
The failing command is:
> /usr/local/bin/autoreconf2.71 -f -i

martinwhitaker, 2023-04-18:
> The failing command is:
> 
> > /usr/local/bin/autoreconf2.71 -f -i

Follow the build instructions (https://steveicarus.github.io/iverilog/usage/installation.html). Run
````
sh autoconf.sh
./configure
make
````

yurivict, 2023-04-18:
This works, thanks!

closed 2023-04-18 by yurivict