steveicarus/iverilog #909: Unable to simulate dff

Sysynth, 2023-04-21:
Hi, 

I am trying to run post-synthesis simulation using Icarus Verilog, but every time I try to simulate something that uses dff cells, I get unknown values ('X') at the outputs.

### Reproduction Steps
Take this simple 4-bit counter as an example:
```
module counter (    input clk,
                    input reset,
                    output reg[3:0] count );

  always @ (posedge clk) begin
    if (reset)
      count <= 0;
    else
      count <= count + 1;
  end
endmodule
```
The synthesized module using sky130 libs (**counter_sky130.v**) looks like this:
```
(* top =  1  *)
(* src = "../../counter.v:1.1-11.10" *)
module counter(clk, reset, count);
  wire _00_;
  wire _01_;
  wire _02_;
  wire _03_;
  wire _04_;
  wire _05_;
  wire _06_;
  wire _07_;
  wire _08_;
  (* src = "../../counter.v:1.27-1.30" *)
  input clk;
  wire clk;
  (* src = "../../counter.v:3.37-3.42" *)
  output [3:0] count;
  wire [3:0] count;
  (* src = "../../counter.v:2.27-2.32" *)
  input reset;
  wire reset;
  sky130_fd_sc_hd__nor2_1 _09_ (
    .A(reset),
    .B(count[0]),
    .Y(_00_)
  );
  sky130_fd_sc_hd__o21bai_1 _10_ (
    .A1(count[0]),
    .A2(count[1]),
    .B1_N(reset),
    .Y(_04_)
  );
  sky130_fd_sc_hd__a21oi_1 _11_ (
    .A1(count[0]),
    .A2(count[1]),
    .B1(_04_),
    .Y(_01_)
  );
  sky130_fd_sc_hd__and3_1 _12_ (
    .A(count[0]),
    .B(count[1]),
    .C(count[2]),
    .X(_05_)
  );
  sky130_fd_sc_hd__a21oi_1 _13_ (
    .A1(count[0]),
    .A2(count[1]),
    .B1(count[2]),
    .Y(_06_)
  );
  sky130_fd_sc_hd__nor3_1 _14_ (
    .A(reset),
    .B(_05_),
    .C(_06_),
    .Y(_02_)
  );
  sky130_fd_sc_hd__and4_1 _15_ (
    .A(count[0]),
    .B(count[3]),
    .C(count[1]),
    .D(count[2]),
    .X(_07_)
  );
  sky130_fd_sc_hd__a31oi_1 _16_ (
    .A1(count[0]),
    .A2(count[1]),
    .A3(count[2]),
    .B1(count[3]),
    .Y(_08_)
  );
  sky130_fd_sc_hd__nor3_1 _17_ (
    .A(reset),
    .B(_07_),
    .C(_08_),
    .Y(_03_)
  );
  (* src = "../../counter.v:5.3-10.6" *)
  sky130_fd_sc_hd__dfxtp_1 _18_ (
    .CLK(clk),
    .D(_00_),
    .Q(count[0])
  );
  (* src = "../../counter.v:5.3-10.6" *)
  sky130_fd_sc_hd__dfxtp_1 _19_ (
    .CLK(clk),
    .D(_01_),
    .Q(count[1])
  );
  (* src = "../../counter.v:5.3-10.6" *)
  sky130_fd_sc_hd__dfxtp_1 _20_ (
    .CLK(clk),
    .D(_02_),
    .Q(count[2])
  );
  (* src = "../../counter.v:5.3-10.6" *)
  sky130_fd_sc_hd__dfxtp_1 _21_ (
    .CLK(clk),
    .D(_03_),
    .Q(count[3])
  );
endmodule
```

**tb_counter.v**
```
module tb_counter;
    reg clk;
    reg reset;
    wire [3:0] count;

    counter count0 (    .clk(clk),
                        .reset(reset),
                        .count(count));

    initial begin
        clk = 1'b0;
        repeat(50) #1 clk = ~clk;
    end

    initial begin
        reset = 1'b1;
        #13 reset = 1'b0;
        #33 reset = 1'b1;
    end

    initial begin
        $monitor("t=%0d, reset=%b, count=%d \n", $time, reset, count);
        $dumpfile("yosys.vcd");
        $dumpvars(0,tb_counter);
    end

endmodule
```
I call iverilog using the following command:
`iverilog -o counter_yy counter_sky130.v tb_counter.v ../../../pdks/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v  ../../../pdks/sky130A/libs.ref/sky130_fd_sc_hd/verilog/primitives.v`

### Expected Behavior
Pre-synthesis simulation:
```
t=0, reset=1, count= x

t=1, reset=1, count= 0

t=13, reset=0, count= 1

t=15, reset=0, count= 2

t=17, reset=0, count= 3

t=19, reset=0, count= 4

t=21, reset=0, count= 5

t=23, reset=0, count= 6

t=25, reset=0, count= 7

t=27, reset=0, count= 8

t=29, reset=0, count= 9

t=31, reset=0, count=10

t=33, reset=0, count=11

t=35, reset=0, count=12

t=37, reset=0, count=13

t=39, reset=0, count=14

t=41, reset=0, count=15

t=43, reset=0, count= 0

t=45, reset=0, count= 1

t=46, reset=1, count= 1

t=47, reset=1, count= 0
```
### Actual Behavior
```
t=0, reset=1, count= x

t=13, reset=0, count= x

t=46, reset=1, count= x
```
I have tried using different synthesis tools and libraries but it does not make a difference.

Am I doing everything correctly when calling iverilog? I am new to the tool so I could be making some mistakes.

Thanks in advance.

martinwhitaker, 2023-04-22:
It looks like you've hit this bug in the sky130 libraries: https://github.com/google/skywater-pdk/issues/310.
Adding -DFUNCTIONAL on the iverilog command line makes your test work.

Sysynth, 2023-04-23:
Thank you so much for the help!

I initially discarded it being a bug with the libraries because I had also tried running the synthesis with global foundries libraries and had the same issue.

Anyways, as you said, adding -DFUNCTIONAL fixes the problem. 😄

Marking this as closed.

closed 2023-04-23 by Sysynth