steveicarus/iverilog #910: Issues with seeing driven signal in VCD (PLI application)

r2com, 2023-04-23:
So I have a simple test design with clk port. My Rather simple PLI C program scans for ports, and then registers necessary callbacks, and one of the callbacks shown below:
```
static PLI_INT32 clk_lo(p_cb_data data)
{
  static int cnt = 0;

  printf("clk_lo called!\n");
  vpiHandle cbH;

  s_cb_data cb;
  s_vpi_value val;
  s_vpi_vecval vecVal;
  s_vpi_time tim, tim2;

  tim.high = 0;
  tim.low = t_delta;
  tim.type = vpiSimTime;

  tim2.high = 0;
  tim2.low = 0;
  tim2.type = vpiSimTime;

  // int value = 1;

  vecVal.aval = 1;
  vecVal.bval = 0;
  val.format = vpiVectorVal;
  val.value.vector = &vecVal;

  vpi_put_value(clk, &val, NULL, vpiNoDelay);

  cb.reason = cbAfterDelay;
  cb.cb_rtn = &clk_hi;
  cb.time = &tim;

  if (cnt < 10) {
    if ((cbH = vpi_register_cb (&cb)) == NULL)
      printf ("cannot register clk_hi call-back\n"); }

    // vpi_free_object(cbH); // make sure you call it to clear cbH

    cnt++;
    return 0;
  }
```

Just calls another callback, and most importantly sets clk port of my DUT with a value 1.

Durign a compile and run, I do not see any issues, the top print statement "clk_lo called!" prints 10 times or so, which is what is planned. In other words, app seems to run fine.

However, my problem is that when I do view VCD dump, I see all top signals of design, but dont see clk being toggled.

Here is what is added in DUT (later I plan to add it to separate top file maybe, but for now it doesnt matter and below code appears at the end of design):
```
initial
 begin
    $dumpfile("test.vcd");
    $dumpvars(0, dut_top);
 end
```

i compile everything and run with:
```
iverilog -o dut_top.vvp -s dut_top -s dump ./src/top.v ./src/dut_dump.v
vvp -M ../vip/build/ -m ../vip/build/libdut_top.vpi.so ./dut_top.vvp
```

So what am I missing? why my signal seems to be driven and PLI C program runs OK, but VCD file not capturing the changes?

r2com, 2023-04-23:
I found that if I change the line in my VPI code for scanning ports to vpiNet instead of vpiPort (not shown in original message)
`iter = vpi_iterate(vpiNet, h_top);`

Then it all works. I wonder why if in above code it is vpiPort it didnt work.. like.. in VPI you cannot drive Ports? just nets? 

anyway.. ill keep issue open for a while just in case if someone has comments, otherwise will move on with what works.

its just for me it was logical that if I scan port and get proper handle for port i should be able to drive port using vpi_put_value, not just net

r2com, 2023-04-23:
I belive ports are hierarchical objects in Verilog, and they are not directly drivable.

martinwhitaker, 2023-04-23:
From the VPI object data model diagrams in the IEEE standard, nets have a `value` property that you can set and get using the `vpi_put_value()` and `vpi_get_value()` functions, but ports do not. So the behaviour you are seeing is correct.

closed 2023-04-23 by martinwhitaker