steveicarus/iverilog #911: getting code coverage with iverilog (question)

r2com, 2023-04-23:
So with ghdl coverage can be done through gcov. 

with iverilog, how would I do that? if I am correct lcov must be used, but what are the steps and what command line arguments must be given to iverilog to produce proper output for coverageprogram?

r2com, 2023-04-23:
I see that one can also use covered, I am also reading some manuals online saying:
_If you are compiling an Icarus Verilog simulation, simply add -m tool_install_root/libexec/covered.vpi covered_vpi.v to the command-line._
however, I cannot se above folder and .vpi files.

So what is the way to go for coverage generation? covered or lcov or what is the right flow?

r2com, 2023-04-25:
Here is a way how I managed to get it working with VCD dump:
```
iverilog -o dut_top.vvp -s dut_top -s dump ./src/top.v ./src/dut_dump.v
vvp -M ../vip/build/ -m ../vip/build/libdut_top.vpi.so ./dut_top.vvp
covered score -v ./src/top.v -t dut_top -i dut_top -vcd test.vcd -o coverage_report.cdd
covered report -d v coverage_report.cdd
```

the "top.v" is the real DUT, and dut_dump.v is just another top level so that VCD dump is created. 
After I ran the sim, I ran above two 'covered' commands and get report with coverage.

However, the original question still holds - how to get it working through the VPI simulation? not through the VCD?

(even though above I did ran VPI sim, but it was just to create dump, the coverage still was through VCD __after__ sim was run)

martinwhitaker, 2023-04-25:
Icarus Verilog has no built-in support for generating coverage statistics. `covered` is the only free tool I know of for doing this. See "Using Covered As a Vpi Module" in the `covered` man page for how to use it via VPI.

closed 2023-04-27 by r2com