steveicarus/iverilog #912: Most effective way to put value in nets using VPI (Question)

r2com, 2023-04-30:
So according to the VPI book, the most effective way to drive some multi bit register with value using **vpi_put_value** is using the format of **vpiVectorVal** and its corresponding **aval** and **bval** fields. The reason for that is the fact that verilog simulators store such registers internally in exactly same type of structure, as a result there is no conversion done on the data passed from vpi_put_value.

Now, I want to confirm, if this is really the case with iverilog? is it also storing and operating internally with reg values in a form of a structure with aval and bval fields? i.e., for example, driving vpi_put_value in an integer format would be slower indeed?

larsclausen, 2023-05-01:
vpiIntVal and vpiVectorVal will have similar performance. vpiIntVal is ever so slightly faster, but I doubt it will be visible above the noise floor.

martinwhitaker, 2023-05-01:
Moving to the discussions section.