steveicarus/iverilog #916: blocking assignments don't seem to be synchronized with nonblocking always blocks in tasks

nlshipp, 2023-05-06:
I'm a newbie to Verilog but I'm seeing a difference between simulations run with iverilog 12 and ISim (14.7) and I think it may be a bug in iverilog.  I'm attempting to simulate a verilog module I'm developing on a Cypress PSOC5LP device and am using their simulation verilog files to test my design.  They've implemented a datapath component with fifos and have written tasks to simulate the CPU reading and writing from the fifo.  When running the test bench with my old version of Xilinx ISim, the read operations work as I expect - the fifo read pointer register is incremented on the falling edge of the cpu_fifo_rd1 signal and the data value is read from the fifo1_data_out wires before the rd_pointer value propagates to the 4 element memory.  However when simulated with iverilog, the rd_pointer register is immediately updated on the rising edge of cpu_fifo_rd1, and the assignment to bitsOut[] from fifo1_data_out has an uninitialized value.  I've put files in github [here.](https://github.com/nlshipp/iverilog-bugs/tree/issue-001)

[isim-001.png](https://github.com/nlshipp/iverilog-bugs/blob/issue-001/isim-001.png) shows the simulation where the rd_pointer is updated on the falling edge, while [iverilog-001.png](https://github.com/nlshipp/iverilog-bugs/blob/issue-001/iverilog-001.PNG) shows that rd_pointer is updated on the rising edge of fifo1_rd_en.

The fifo0_read and fifo1_read tasks using blocking assignments are in the cy_psoc3_dp.v file along with the fifo simulation component DP_FIFO that uses non-blocking assignments.

nlshipp, 2023-05-28:
I've added a simplified test bench that shows part of the underlying issue I think I'm seeing.  [test_inc_TB.v](https://github.com/nlshipp/iverilog-bugs/blob/issue-001/inc_test_TB.v)
I've implemented a module that increments an internal counter on the positive edge of a clock if the increment wire is asserted.  In the test bench I have two instances of the module, one is driven by the test bench using blocking assignments (test_inc1), and the other is driven by an always statement using non-blocking assignments (test_inc2).  In the output, the blocking assignments increment the counter twice - once on the rising edge of test_inc1 and once again on the falling edge of test_inc1, while the non-blocking assignments only increment the counter once and delayed by one clock cycle.

For the double increment, why is inc being set to one on the falling edge of test_inc1?  The component has

```
	always @(posedge clk)
	begin
		if (test_inc)
		begin
			inc <= 1;
			@(negedge clk);
			inc <= 0;
		end
		else
			inc <= 0;
	end
```
So why would it set inc <= 1 on both @(posedge clk) ; test_inc1 = 1, and @(posedge clk); test_inc1 = 0?

[trace image](https://github.com/nlshipp/iverilog-bugs/blob/issue-001/inc_test-001.png)

Again, simulating this using my old version of Xilinx ISim, the simulation shows both blocking and non-blocking paths have identical behaviour.

caryr, 2023-06-21:
The non-blocking code has a race. Specifically which posedge is executed first? The one in the inc_test module or the one in the test bench. The one in the test bench executes first at T=12 which allows test_inc to be set before the posedge in the inc_test module so it increments at 12. At 20 the posedge in the inc_test module executes first which allows a second increment before test_inc is dropped in the TB.

Like @martinwhitaker said when there is a difference between block and non-blocking it is almost always because of a race. Learning to figure out these races is a critical skill in becoming proficient in SystemVerilog. Icarus makes this a little easier since you can annotate the VVP code to emit file/line information as the procedural commands are executed. This is done by using the `-pfileline=1` flag when compiling to add the instrumentation to the compile output. You then want to have a `$stop` before you want to enable the tracing. From the interactive prompt type `trace on` and then `cont`. You could have a second `$stop` after the time you want to debug and type `trace off` followed by `cont` to only print the verbose debug information over a given simulation time.

caryr, 2023-06-21:
For example:

```
                @(negedge clk);
                $stop; // Type 'trace on' followed by 'cont'
                @(posedge clk);
                test_inc1 = 1'b1;
                @(posedge clk);
                test_inc1 = 1'b0;
                @(negedge clk);
                $stop; // Type 'trace off' followed by 'cont'
                #40;
```

nlshipp, 2023-06-27:
I think I understand a bit better now, but @caryr , isn't it the blocking code that is introducing the race?  What I see at T=12, is that we have a clock tick with multiple statements waiting for the posedge of clk.  For the non-blocking assignment of test_inc2 to 1, it's scheduled.  However for the blocking assignment of test_inc1 to 1, it updates the state immediately before the instance of inc_test that's connected to test_inc1 has a chance to execute.  The reverse is true at T=20, both instances of inc_test are processed before the blocking assignment of test_inc2 back to zero is performed which causes the 2nd increment.

So the difference I see between the Icarus and Xilinx Isim is that for ISim, it's more forgiving of mixing blocking and nonblocking assignments.  Perhaps at each simulation event, it prioritizes and schedules non-blocking code simulation first and then blocking statements second.  While Icarus, I wouldn't say it's random, but at T=12, it ran the blocking assignment ``test_inc = 1;`` at line 31 first before running the inc_test modules, at T= 20, it ran the inc_test modules first before running the blocking assignment ``test_inc1 = 0;`` at line 33.  

If I put the test_inc1 assignments inside a for loop:
```
	for (integer i = 0 ; i < 8 ; i = i + 1)
	begin
		@(posedge clk);
		test_inc1 = 1'b1;
		@(posedge clk);
		test_inc1 = 1'b0;
	end
```
then even though test_inc1 is regular square wave, the statements in inc_test only ever see the value of it as 1 for the first two clock cycles.  After that point, the simulator settles down and always schedules the inc_test code to run before the assignment to 1, and always after the assignment to 0, so inc_test only ever sees the value as 0 and out1 is never updated again regardless of the number of iterations the loop runs.

I can resolve the issue by adding a zero length delay before the assignment of test_inc1 to force the assignment to be done after all the other non-blocking statements are simulated, but is that an accepted practice?
```
	@(posedge clk);
	#0 test_inc1 = 1'b1;
	@(posedge clk);
	#0 test_inc1 = 1'b0;
```

I've updated my test bench to show both behaviours [inc_test_TB.v](https://github.com/nlshipp/iverilog-bugs/commit/c8ecb10bdd554b93811fa04c27899da5b7249d3a)

Thank you for the instrumentation and trace enabling tips.  They were very useful to see what was happening in the simulator.

martinwhitaker, 2024-01-09:
When there are multiple events scheduled at the same time, a simulator is free to execute them in any order. Different simulators implement the scheduling algorithm in different ways, so you can expect different behaviour. To learn more about this, read the Scheduling Semantics section in the LRM (IEEE 1364-2005 for traditional Verilog, IEEE 1800-2017 for SystemVerilog, both of which are freely downloadable from the IEEE website for personal use). I recommend starting with 1364-2005 - the SystemVerilog scheduling semantics are more complicated.

Using delays to work round races is generally not a good idea - better to use non-blocking delays everywhere, or to change values on one edge of the clock and sample them on the other edge.

Closing as this doesn't look like a bug.

closed 2024-01-09 by martinwhitaker