steveicarus/iverilog #925: Unable to get version

trayres, 2023-05-22:
Compiling from source w/ MSYS2 Mingw64, version s20221226-127-gdeeac2edf, on Windows 10 x64, I am getting an error when asking iverilog for its version:

`$ iverilog -V
Icarus Verilog version 13.0 (devel) (s20221226-127-gdeeac2edf)

Copyright (c) 2000-2021 Stephen Williams (steve@icarus.com)

  This program is free software; you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation; either version 2 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License along
  with this program; if not, write to the Free Software Foundation, Inc.,
  51 Franklin Street, Fifth Floor, Boston, MA 02110-1301, USA.

Icarus Verilog Preprocessor version 13.0 (devel) (s20221226-127-gdeeac2edf)

Copyright (c) 1999-2021 Stephen Williams (steve@icarus.com)

  This program is free software; you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation; either version 2 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License along
  with this program; if not, write to the Free Software Foundation, Inc.,
  51 Franklin Street, Fifth Floor, Boston, MA 02110-1301, USA.
Unable to get version from "C:\msys64\mingw64\lib\ivl\ivl -V -C"C:\msys64\tmp\ivrlh1166" -C"C:\msys64\mingw64\lib\ivl\vvp.conf""`

I believe the "Unable to get version" is causing all my Yosys compilation tests to fail, as the tests check for the current Git version before continuing.

martinwhitaker, 2023-06-05:
I am unable to reproduce this with a clean build and install from the master branch, using a fully up-to-date version of MSYS2 /MINGW64 in Windows 10:
````
$ which iverilog
/mingw64/bin/iverilog

$ iverilog -V
Icarus Verilog version 13.0 (devel) (s20221226-153-gc74048a52)

Copyright (c) 2000-2021 Stephen Williams (steve@icarus.com)

[snip license]

Icarus Verilog Preprocessor version 13.0 (devel) (s20221226-153-gc74048a52)

Copyright (c) 1999-2021 Stephen Williams (steve@icarus.com)

[snip license]

Icarus Verilog Parser/Elaborator version 13.0 (devel) (s20221226-153-gc74048a52)

Copyright (c) 1998-2022 Stephen Williams (steve@icarus.com)

[snip license]

 FLAGS DLL vvp.tgt
vvp.tgt: Icarus Verilog VVP Code Generator 13.0 (devel) (s20221226-153-gc74048a52)

Copyright (c) 2001-2021 Stephen Williams (steve@icarus.com)

[snip license]
````

caryr, 2023-06-15:
Were you able to resolve this?

caryr, 2023-09-04:
No more information provided so closing this report.

closed 2023-09-04 by caryr