steveicarus/iverilog #928: implicitly importing a package , iverilog cannot find declarations in package.

Xhopo, 2023-06-02:
pkt as follows:

```
`timescale 1ps/1ps
package test_pkt;
    parameter A = 3'd0;
    parameter B = 3'd2;
    function [3:0] add;
        input [3:0] a;
        input [3:0] b;

        add = a+b;
    endfunction
endpackage
```

test.sv as follows:
```
import test_pkt::*;

`timescale 1ps/1ps
module test;

reg[3:0] to;

initial begin
    to = add(A,B);
    # 100ns
    $finish();
end
    
endmodule
```
compile with:

`iverilog -g2012 *.sv`

ERROR results:
```
test.sv:9: error: No function named `add' found in this context (test).
1 error(s) during elaboration.
```

Xhopo, 2023-06-02:
Explicitly specifying the method context, following compile pass:

```
import test_pkt::*;
`timescale 1ps/1ps
module test;
reg[3:0] to;
initial begin
    to = test_pkt::add(test_pkt::A,test_pkt::B);  // Add Package Name
    # 100ns
    $finish();
end
endmodule
```

martinwhitaker, 2023-06-03:
You can work around this bug by moving the import statement to inside the test module.

Xhopo, 2023-06-05:
It works now, and the internal import solved this problem. Thanks.

PR #938 (2023-06-11): Handle implicit task/function imports in the unit scope 
SystemVerilog requires that functions and tasks are not implicitly imported
if a symbol with the same name appears in the scope, even if it the symbol
is declared after its usage.

To support this a list of potential imports is collected while parsing a
scope and only when the end of the scope is reached it is evaluated whether
the symbol should be imported or not based on whether it already exists in
the scope.

This currently works fine for all scopes except for the unit scope. Since
the unit scope might span multiple files it is never [... cut]
closed 2023-06-05 by Xhopo