steveicarus/iverilog #933: Help Wanted: no output file for vvp generated in powershell

zhang-stephen, 2023-06-11:
ENV:
 - Windows 11 23H2
 - Powershell 7.2
 - iverilog 11.0, installed by msys2

I am playing with iverilog for some simple simulation. And I am trying the example provided [here](https://steveicarus.github.io/iverilog/usage/gtkwave.html).

Then I got nothing when I ran `iverilog -o dsn counter_tb.v counter.v` in powershell, but got valid output in msys2 shell.

Is the powershell or Command Prompt not supported yet? Any suggestions?

caryr, 2023-06-11:
We do not provide Windows binaries and have limited experience on Windows in general so I'm not certain how much we can help. In the past I have used cygwin to compile and run on Windows. WSL should work, but I'm not certain it will support GTKWave.

martinwhitaker, 2023-06-11:
Last time I checked, it worked when using the Windows Command Prompt. But you need to make sure the MSYS2 DLLs used by iverilog are on the Windows search path or have been copied into the same directory as iverilog.exe.

caryr, 2023-06-15:
Were you able to resolve this issue? This appears to be a configuration/setup issue and not a bug in Icarus.

zhang-stephen, 2023-06-24:
I cannot got correct output in powershell yet, but it works well in cmd.

caryr, 2023-09-04:
No more information provided so closing this report.

closed 2023-09-04 by caryr

DvvCz, 2023-10-26:
Yeah seems WSL does not give any output either. I downloaded through `sudo apt install iverilog` and `vvp` said it was version 11.

Works fine with regular windows terminal (v12 - https://bleyer.org/icarus), just an inconvenience since I'm trying to test my script to use icarus verilog for continuous integration on github.

Attaching necessary files and output in case you want to pursue this, specific (system) verilog code likely does not matter.

``src/FLC.sv``
```verilog
module FLC (input FLC_A, input FLC_B, input FLC_C, output FLC_Q);
	assign FLC_Q = (FLC_B & FLC_C) | ~(FLC_A | FLC_B);
endmodule
```

``tests/FLC_TB.sv``
```verilog
`timescale 1ns / 1ps

`include "../src/FLC.sv"

module FLC_TB ();

	reg flc_a;
	reg flc_b;
	reg flc_c;

	wire flc_q;

	FLC dut (
		.FLC_A (flc_a),
		.FLC_B (flc_b),
		.FLC_C (flc_c),
		.FLC_Q (flc_q)
	);

	initial begin
		flc_a = 0; flc_b = 0; flc_c = 0;
		#10 if (flc_q != 1) $error("0, 0, 0 != 1");

		flc_a = 0; flc_b = 0; flc_c = 1;
		#10 if (flc_q != 1) $error("0, 0, 1 != 1");

		flc_a = 0; flc_b = 1; flc_c = 0;
		#10 if (flc_q != 0) $error("0, 1, 0 != 0");

		flc_a = 0; flc_b = 1; flc_c = 1;
		#10 if (flc_q != 1) $error("0, 1, 1 != 1");

		flc_a = 1; flc_b = 0; flc_c = 0;
		#10 if (flc_q != 0) $error("1, 0, 0 != 0");

		flc_a = 1; flc_b = 0; flc_c = 1;
		#10 if (flc_q != 0) $error("1, 0, 1 != 0");

		flc_a = 1; flc_b = 1; flc_c = 0;
		#10 if (flc_q != 0) $error("1, 1, 0 != 0");

		flc_a = 1; flc_b = 1; flc_c = 1;
		#10 if (flc_q != 1) $error("1, 1, 1 != 1");

		$finish;
	end

endmodule
```

```
iverilog -g2012 -Isrc -o temp tests/FLC_TB.sv
```

`temp`
[... 283 lines cut ...]
T_0.12 ;
    %pushi/vec4 1, 0, 1;
    %store/vec4 v000001f35580f8f0_0, 0, 1;
    %pushi/vec4 1, 0, 1;
    %store/vec4 v000001f35580f990_0, 0, 1;
    %pushi/vec4 1, 0, 1;
    %store/vec4 v000001f35580fa30_0, 0, 1;
    %delay 10000, 0;
    %load/vec4 v000001f35580fad0_0;
    %pad/u 32;
    %cmpi/ne 1, 0, 32;
    %jmp/0xz  T_0.14, 4;
    %vpi_call/w 3 43 "$error", "1, 1, 1 != 1" {0 0 0};
T_0.14 ;
    %vpi_call/w 3 45 "$finish" {0 0 0};
    %end;
    .thread T_0;
# The file index is used to find the file name in the following table.
:file_names 5;
    "N/A";
    "<interactive>";
    "-";
    "tests/FLC_TB.sv";
    "src/../src/FLC.sv";
```