steveicarus/iverilog #941: trying simulation with xilinx IP.

bansalrajesh, 2023-06-13:
Hi All,
  I am using icarus version 12 on ubuntu. I am trying to compile unisim and compile netlist of a xilinx fifo IP.
I get this error. around 101 unresolved items.

unresolved vvp_net reference: v0x5643f7a05b70_0
compile_cleanup: 101 unresolved items

here is the make file commands.

VERILOG_INCLUDE_DIRS += $(XILINX_PATH)/src/
VERILOG_INCLUDE_DIRS += $(XILINX_PATH)/src/unisims/
VERILOG_INCLUDE_DIRS += $(RTL_ROOT)/vivado/fifo.srcs/sources_1/ip/axis_data_fifo_0/hdl/

ifdef VERILOG_INCLUDE_DIRS
    COMPILE_ARGS += $(addprefix -I, $(VERILOG_INCLUDE_DIRS))
endif

COMPILE_ARGS += -s dac_memory_top_tb -s glbl
COMPILE_ARGS += -g2012

any help on this is highly appreciated.

Best

caryr, 2023-06-15:
There are a couple ways to figure this out. You can remove or stub out pieces of the code until you find the code that is causing the problem. Since this looks like the compiler is producing incorrect code we can try to backtrack from the generated code what RTL code is causing the problem. The unresolved vvp_net is the likely culprit.

caryr, 2023-09-04:
No more information provided so closing this report.

closed 2023-09-04 by caryr