steveicarus/iverilog #943: [SDF] Tracking Issue

mole99, 2023-06-14:
This issue tracks all open issues regarding Standard Delay Format (SDF) support in Icarus Verilog. Additionally, issues for `specify` are listed as they are often related.

**SDF**

- [ ] #746
- [ ] #509
- [ ] #899

**Specify**

- [ ] #107

#### Source Forge Issues:

**SDF**

- SDF back annotation fails/aborts for a tri-state buffer https://sourceforge.net/p/iverilog/bugs/960/
- No support for annotating specparams from SDF file https://sourceforge.net/p/iverilog/bugs/889/

**Specify**

- specify block delays break when multiple inputs have the same driver https://sourceforge.net/p/iverilog/bugs/954/
- Unexpected race of modpath "if" expressions (pr1877743.v) https://sourceforge.net/p/iverilog/bugs/495/
- Path delay expression without parenthesis not supported https://sourceforge.net/p/iverilog/bugs/493/
- Module path delays do not properly filter pulses https://sourceforge.net/p/iverilog/bugs/888/

caryr, 2023-06-15:
For the pieces that are not implemented or working correctly you may want to consider parsing them and ignoring them after emitting a warning either in the RTL parser or the SDF parser. You can then get the compiler to handle the RTL part correctly and have the targets emit the warning or do the correct thing.

There is also the implementation of path delays which is needed for SDF interconnect delays. I think there is an old PR for that.

mole99, 2023-06-15:
Yes, that's what I would like to do for every unsupported feature. If you know any unsupported feature that needs a warning just tell me and I will add it.

Do you maybe recall what this PR was named? I searched with the keywords `path`, `delay` and `interconnect` but could not find anything.

caryr, 2023-06-15:
Here are some other SF PRs:

https://sourceforge.net/p/iverilog/bugs/740/ - timing checks
https://sourceforge.net/p/iverilog/bugs/273/ - wire delays

There are also a couple other PRs that will be fixed when SF-495 is fixed. I can help you track those down later.

mole99, 2023-06-16:
Thanks for the issues, I have added them to the list.

It seems we have a different understanding of what a PR is :)
Pull Request (PR) is GitHub jargon for when you want to add new features to a branch.
If you are using GitLab, it is called Merge Request (MR) there.

See here for a better explanation: [pull-request-vs-merge-request](https://stackoverflow.com/questions/22199432/pull-request-vs-merge-request)

caryr, 2023-06-16:
If you notice many of the tests start or have pr in their title. I believe this is done to indicate the problem report they where initially ties to. We added gh to indicate they are from github instead of SourceForge. It's sometimes useful when we need to track a change back to an initial request. Of course it didn't help when SF reordered the reports to be project specific.

mole99, 2023-06-16:
Oh, I see! I have wondered why they are named that way ^^

martinwhitaker, 2024-01-08:
Would it be better to add a new "SDF" label rather than have a tracking issue? You could then use the filter expression "label:SDF" to find all the relevant issues.

mole99, 2024-01-09:
That's a great idea! But I don't think I have the permission to create and assign labels. Maybe someone else could do that?

caryr, 2024-01-09:
I added a specify/SDF tag. I marked the issues on the first page using this tag. We need to mark the rest and then this can be closed except this report does still have some benefit since it references the SourceForge bug reports.