steveicarus/iverilog #955: Running demo on instructions doesn't work when using command file

IrishMedicalStudent, 2023-06-18:
The demo here (https://steveicarus.github.io/iverilog/usage/getting_started.html#) doesn't appear to work? When the attached files are used and the commands:

`iverilog -o my_design -c File_List.txt`
`vvp my_design`

Are ran, there is no output. Running:

`iverilog -o Counter_TB counter_tb.v counter.v `
`vvp Counter_TB`

Works fine though? The my_design command seems to generate a truncated version of the expected result.

[Samples.zip](https://github.com/steveicarus/iverilog/files/11780293/Samples.zip)

caryr, 2023-06-20:
What system and version of iverilog are you using? This works as expected on Ubuntu 22.04 using the latest development version of Icarus.

hlutor, 2023-07-26:
I also experienced weird behavior when using a command file in a similar manner. This was while using the v11-branch on Ubuntu 20.04. Updating to the v12-branch fixed it.

caryr, 2023-09-04:
No more information provided so closing this report.

closed 2023-09-04 by caryr