steveicarus/iverilog #958: task prosedure problem

balanx, 2023-06-20:
msys2
Icarus Verilog version 12.0 (devel) (s20150603-1391-gd480c4d7d)

$ iverilog -g2001 task_test.v

```

module task_test;

reg  x = 0;
reg  [7:0] y = 0;

reg  clk = 0;
always #5 clk  = ~clk;

initial begin
    $dumpvars;
    #100;
    repeat(10) tt(x, y);
    #100 $finish;
end

task tt;
inout  a;
inout [7:0] b;

begin
    @(posedge clk) a = ~a;
    @(negedge clk) b = b + 1;
end
endtask

endmodule

```

The result is the below,
(sorry, i can't upload gtkwave pic.)

My question is that the signals 'x' and 'y'  always change at the same negedge, but I descripted 'x' is posedge and 'y' is negedg.

balanx, 2023-06-20:
[image]
this is the simulation wave.

caryr, 2023-06-20:
You misunderstand how task arguments are passed. When a task is called the internal input and inout values are assigned from the argument list and it is only on exit of the task that the internal values are assigned to the inout and output arguments.

If you look the values internal to the task are changing at the correct time. The simulator does not current support the ref argument type which could be used to make this work as you are expecting. You can also use the x and y values directly in the task to avoid passing them and the assignment restrictions that come with that, but that makes the task specific to those two external variables.

As written this is expected behavior (output and inout variables are being written when the task exits).

balanx, 2023-06-22:
@caryr Thanks for your reply, You are right. I appriciate your explaination.

closed 2023-06-22 by balanx