steveicarus/iverilog #959: Is "automatic logic" inside a "function automatic logic" module supported

UweBonnes, 2023-06-21:


caryr, 2023-06-21:
I believe at the moment we only support the whole function being automatic or not. For this case where the entire function is already automatic the declaration is superfluous so we could just read the `automatic` attribute and ignore it, but it looks like we have not done that yet.

caryr, 2023-06-21:
I build V12 on my development machine and it also works so you must be using an outdated version of Icarus. Please confirm that updating to V12 or newer solves your problems.

UweBonnes, 2023-06-21:
Sorry, I complied from GIT master, but somehow managed to pick up the Opensuse Leap15.4 available iverilog version Version  10.1. It works with the GIT version.
B.t.w. What about always printing the iverilog version when some error is thrown?

closed 2023-06-21 by UweBonnes