steveicarus/iverilog #960: Does iverilog support "int unsigned" of SystemVerilog?

vicg42, 2023-06-24:
I tried to simulate [cvfpu](https://github.com/openhwgroup/cvfpu)(a floating-point unit written in SystemVerilog) with iverilog, but the compiler gave up every time when there was parameter int unsigned in source code even if I added -g2005-sv. Does it mean that int unsigned is not supported by iverilog? And I'll also be appreciated that you can tell me how to simulate SystemVerilog codes, like cvfpu, with iverilog (not vcs).

(https://stackoverflow.com/questions/75888393/does-iverilog-support-int-unsigned-of-systemverilog)

larsclausen, 2023-06-24:
This is supported in the latest version of iverilog.

caryr, 2023-06-24:
I tried the simple example from the discussion on stackoverflow:
```
module top;
  parameter int unsigned DATA = 3;
  initial $display(DATA);
endmodule
```
and this fails in both V12 and development with the following:
```
gh960.v:2: error: parameter declared outside parameter port list must have a default value.
gh960.v:2: syntax error
gh960.v:2: error: Invalid module item.
```
Something is certainly getting parsed incorrectly.

It's also worth noting that for some reason EDA playground has not updated the Icarus version in a very long time. The latest they have installed is a development version of 10.0 and we released 10.0, 10.1, 10.2, 10.3, 11.0 and 12.0 over the intervening years.

Maybe you can request they update `iverilog` to a newer version to benefit the community at large.

mithro, 2023-06-24:
M
3K97 7N0PA9J90M

larsclausen, 2023-06-24:
> I tried the simple example from the discussion on stackoverflow:
> 
> ```
> module top;
>   parameter int unsigned DATA = 3;
>   initial $display(DATA);
> endmodule
> ```
> 
> and this fails in both V12 and development with the following:

 It works in SystemVerilog mode
```
$ iverilog -g2005-sv int_parameter.v
$  ./a.out
        10
```

caryr, 2023-06-25:
Yes, you are correct. I thought I was using an alias, but I had not set that up in my new VNC. This does work as expected in both V12 and development.

caryr, 2023-09-04:
No more information provided so closing this report.

closed 2023-09-04 by caryr