steveicarus/iverilog #962: Execution does not stop after $finish call

AndreasLoow, 2023-06-26:
Hi, I have noticed that Icarus seems to finish the current simulation cycle instead of exiting immediately after a call to `$finish`. E.g., consider the following module illustrating this:

```systemverilog
module test;

logic x = 1;

initial begin
 $monitor("monitor: ", x);
 $finish(x);
end

initial x = 0;

// still happens with the following variants:
//
// initial x <= 0;
// initial #0 x = 0;
// initial x = #0 0;
//
// but not with longer delays:
//
// initial #1 x = 0;
// initial x = #1 0;

initial $display("display: ", x);

endmodule
```

The output from running the above is:

```
> iverilog test.sv && ./a.out
test.sv:7: $finish(1) called at 0 (1s)
display: 0
monitor: 0
```

The standard just says the following about `$finish`:

> The `$finish` system task causes the simulator to exit and pass control back to the host operating system.

It seems that at least the four commercial simulators on EDA Playground interpret this as "exit immediately" rather than "finish the current simulation cycle and then finish".

Since the standard does not say much it is open to interpretation, but is it possible that "exit immediately" is a more convincing read than "finish what you are doing and then exit"?

E.g., with the "finish what you are doing and then exit" interpretation, we can have long/infinite executions after a `$finish$` call and delay the exit/never actually exit:

```systemverilog
module test2;

initial $finish;

initial forever;

endmodule
```

caryr, 2023-06-26:
It's a little more complicated since you still need to execute final blocks, but the non-blocking and #0 delays should probably not be executed.

A long time ago I wrote a checker in the compiler to look for infinite loops, but this may have only been for always constructs, so it should likely be enhanced to handle statements in initial blocks as well since what you wrote obviously creates an infinite loop.

AndreasLoow, 2023-06-26:
Ah, good point, yeah, final blocks definitely need to be executed... A little less clear what to do about monitors, seems that other simulators do not execute installed monitors after $finish calls.

(Btw, re checking initial blocks: final blocks can loop forever as well.)

red0bear, 2023-06-26:
A finish must be always in some edge of some signal with some conditions to meet the end of simulation.

AndreasLoow, 2023-06-27:
Not sure what you mean? Here is one example of a final-based module that currently runs forever without a warning:

```systemverilog
module test;
   
final forever;

endmodule
```

red0bear, 2023-06-27:
You need a understand computer program logic is quite diff from verilog behavior logic. If you wish to dig ... you can look at my projects and try figure out.