steveicarus/iverilog #972: include file not found

holography-be, 2023-07-16:
Hello,
Include file is not found during verify.

I use apio as tool to verify and build the bitstream. (win10).

First line of one of v file is
`include "C:\Users\***\Documents\PlatformIO\Projects\StateMachine\include\updown.v"
got error from iverilog (launch with apio verify)
iverilog -o hardware.out -D VCD_OUTPUT= -D NO_ICE40_DEFAULT_ASSIGNMENTS "C:\Users\***\.apio\packages\tools-oss-cad-suite\share\yosys/ice40/cells_sim.v" clockDivider.v debouncer.v statemachine.v timer.v
statemachine.v:2: Include file C:\Users\***\Documents\PlatformIO\Projects\StateMachine\include\updown.v not found

Do I missed something ?

Thank you.

mole99, 2023-07-18:
Hi @holography-be, does the file `C:\Users\myName\Documents\PlatformIO\Projects\StateMachine\include\updown.v` exist on your computer?

mole99, 2023-07-18:
Good to hear that your issue is solved!

Maybe you could try to use [openFPGALoader](https://github.com/trabucayre/openFPGALoader) to upload your bitstream to the board (if you don't want to use Diamond).

You can report your bug at [apio](https://github.com/FPGAwars/apio) so that it may be solved.

If your issue here is solved you can click on "Close issue". Thanks!

holography-be, 2023-07-19:
Thanks.
I'll take a look to openFPGALoader.

caryr, 2023-08-10:
I assume this is not a problem in Icarus so can this be closed?

closed 2023-09-04 by caryr