steveicarus/iverilog #974: v12/v13: ERROR Port declaration inherits dimensions from var/net

vmw, 2023-07-19:
Hi,

When compiling an altera simulation file, using iverilog versions 12 or the latest commit on master, and using our internal test benches, we get a number of errors similar to the following:

```
altera_rtl/simulation/altera_mf.v:3704: warning: Port declaration of E0_mode inherits dimensions from var/net.
altera_rtl/simulation/altera_mf.v:3704: error: Scalar port ``E0_mode'' has a vectored net declaration [48:1].
altera_mf.v:3704: warning: Port declaration of E1_mode inherits dimensions from var/net.
altera_rtl/simulation/altera_mf.v:3704: error: Scalar port ``E1_mode'' has a vectored net declaration [48:1].
altera_rtl/simulation/altera_mf.v:3704: warning: Port declaration of E2_mode inherits dimensions from var/net.

```

The specific code that causes the error looks like this:

```
    reg [8*6:1] L0_mode, L1_mode, G0_mode, G1_mode, G2_mode, G3_mode, E0_mode, E1_mode, E2_mode, E3_mode;
    reg [8*5:1] scan_chain;
    reg [31:0] m_mod;
```

We've determined that this test bench passes when we use iverilog versions from the following branches:

v10-branch (iverilog-git-v10_3.r42.gb98854309-1-x86_64.pkg)
v11-branch (iverilog-git-v11_0.r132.gb2f2414f4-1-x86_64.pkg)

But, using the latest released version, and also using the development nightly build, it fails with the errors mentioned above.

Reviewing previous issues, it looks like this may be a regression, as it was previously reported here, and marked as fixed: https://github.com/steveicarus/iverilog/issues/253

larsclausen, 2023-07-23:
You can turn the error into a warning by passing the `-gno-io-range-error` flag when running `iverilog`. See #788.

vmw, 2023-07-24:
Thank you very much! I can confirm that the flag works to revert back to the usual behaviour.

closed 2023-07-24 by vmw