steveicarus/iverilog #977: allow localparam in parameter port list

galenseitz, 2023-07-25:
Hi,

With following code, iverilog (v12.0 built from git under CentOS 7.9) generates a syntax error on the line containing localparam. It would be quite useful if iverilog were to support the use of localparam in the parameter port list.

`
module foo
  #(parameter
    SIZE = 256
    localparam BITS = $clog2(SIZE)
    )
   (
    input [BITS-1:0]	 a,
    output [BITS-1:0] y
//    input [7:0]	 a,
//    output [7:0] y
    );

   assign y = a;

endmodule
`
In my search of the iverilog github issues, I didn't find anything related to this particular use of localparam. The commercial simulators on EDA Playground all seem to support this.

Thanks.

larsclausen, 2023-07-26:
Hi,

This is supported in the v12.0 release. In your example there is a comma missing before the `localparam`. If you add it it should work.

galenseitz, 2023-07-26:
Doh! Sorry about that. I used valid code for my experiments on EDA playground, but version 12 isn't available there, or I would have caught my mistake. Thanks for the response!

closed 2023-07-26 by galenseitz