steveicarus/iverilog #978: Unexpected change in stdout chatter between 11.0 and 12.0

ldoolitt, 2023-07-28:
I accidentally (!) ran some of my Verilog test suites recently under Debian testing (Trixie).  This includes iverilog 12.0, where I normally use iverilog 11.0 found in Debian stable (bookworm).
I got unexpected failures due to differences in the stdout chatter between iverilog 11.0 and 12.0.

To reproduce, first create two files
```
cat > basal.v << EOT
module test;
initial begin
    \$display("Hello", 2+2);
\`ifdef FIN
    \$finish();
\`endif
end
endmodule
EOT
cat > bb.sh << EOT
iverilog -V | awk 'FNR==1{print \$0}'
iverilog -Wall -o basal basal.v && vvp basal
iverilog -DFIN -Wall -o basal basal.v && vvp basal
echo "--"
EOT
```
Here are my results on Debian stable
```
$ sh -x bb.sh
+ iverilog -V
+ awk FNR==1{print $0}
Icarus Verilog version 11.0 (stable) ()
+ iverilog -Wall -o basal basal.v
+ vvp basal
Hello          4
+ iverilog -DFIN -Wall -o basal basal.v
+ vvp basal
Hello          4
+ echo --
--
```
and Debian testing
```
$ sh -x bb.sh
+ iverilog -V
+ awk FNR==1{print $0}
Icarus Verilog version 12.0 (stable) ()
+ iverilog -Wall -o basal basal.v
+ vvp basal
Hello          4
+ iverilog -DFIN -Wall -o basal basal.v
+ vvp basal
Hello          4
basal.v:5: $finish called at 0 (1s)
+ echo --
--
```
See the extra "basal.v:5: $finish called at 0 (1s)"?
Any test bench that captures stdout from such a Verilog program is potentially sensitive to this version change.  That includes the simple case of comparing the output against a gold file.

In some cases I can and probably will rearrange my test benches so the Verilog program calls $fwrite() instead of just $display(), manage the output file name with plusargs, and ignore the stdout chatter.  But not every use-case can be shoehorned into that paradigm.

Assuming this change in default behavior is intentional, is there a (backwards-compatible) way to turn it off?  I expect Users of both iverilog versions to be able to run my regression tests, and I'm not a fan of the complexity rabbit-hole of autoconf-style detection of program versions.

Obligatory xkcd: https://xkcd.com/1172/

larsclausen, 2023-07-28:
The 12.0 behavior is the behavior as required by standard. If `$finish` is called with no argument it is supposed to print the simulation time and file location.

You can get the same behavior in both 11.0 and 12.0 by calling `$finish(0)`. This will suppress any diagnostic output. See #449

ldoolitt, 2023-07-28:
> You can get the same behavior in both 11.0 and 12.0 by calling `$finish(0)`.

Confirmed. That's really helpful, thanks!
Other than needing to use a sed script to modify 113 files in my code base, that's an easy fix.

closed 2023-07-28 by ldoolitt