steveicarus/iverilog #987: Crash in VPI module initialisation.

gatk555, 2023-08-11:
This is probably not really an Icarus issue, but I would appreciate any comments.

I have a VPI module that calls the Gtk graphics library. Following an OS upgrade to Debian 12, I am seeing intermittent behaviour like this:

iverilog -o test2 -m ./panel.vpi test2.v
Segmentation fault (core dumped)

The problem occurs with both the OS-packaged Icarus (Version 11) and a build from today's git source.  The crash seems to be happening in g_thread_new() from the Glib library, part of Gtk's underpinnings.  So this appears to be a GNOME problem, and I should try building a debug version of Glib to dig deeper.

However, that looks like heavy work.  I have been using this code for years without problems, so I suspect there has been some relevant change to Icarus that is triggering the bug, probably introduced in version 11.  (But it might be a Glib change.)  Any suggestions would be helpful.

In the meantime, I work around the problem with

IVL=IVL verilog -o test2 -m ./panel.vpi test2.v

and a call to getenv() that detects compilation rather than simulation and skips the Gtk and Glib calls.

Environment: Debian 12.1 on X86-64.

red0bear, 2023-08-11:
:quite possible you doing something weird , VPI under C work on certain point ... but it can crashes for many reasons.

caryr, 2023-08-11:
With the OS update did Icarus also get updated? If so try the earlier Icarus version to see if the issue goes away. If the version was not updated or the earlier version has the same issue then you will need to do the heavy lifting to figure out why Gtk is failing. Which is the likely case. I have no idea why GNOME would fail. Have you search to see if these kind of failures are because of some configuration change/bug in GNOME, etc.

gatk555, 2023-08-13:
Weird?  Surely everyone wants a GUI panel on their simulations?  Very handy for early testing and debugging.  The code is very straightforward and has given little trouble for years.

> With the OS update did Icarus also get updated? 

Good question.  It seems I was guilty of an assumption, as on checking the package repository, both Debian 11 and 12 come with iverilog 11.  I am here on a fishing trip but have caught nothing, so closing.

Thanks for the comments.

closed 2023-08-13 by gatk555

red0bear, 2023-08-13:
GUI gives a lot of extra work. I hope you get well paid to do that.

gatk555, 2023-08-14:
No pay and little work: about 3300 lines of code over 10 years.  Sample output can be seen in the README of my "blink" repository.