steveicarus/iverilog #988: How can I return a 64-bit vector through function using VPI?

Devil-SX, 2023-08-14:
I have declared `.sizetf` 

```C
void register_systolic_array_vpi() {
    s_vpi_systf_data *tf_data_ptr = (s_vpi_systf_data *)malloc(sizeof(s_vpi_systf_data));
    if (!tf_data_ptr) {
        // Handle memory allocation failure
        return;
    }

    // Register $initialize_matrix
    *tf_data_ptr = (s_vpi_systf_data){
        .type           = vpiSysTask,
        .sysfunctype    = vpiSysTask,
        .tfname         = "$initialize_matrix",
        .calltf         = initialize_matrix_calltf,
        .compiletf      = NULL,
        .sizetf         = NULL,
        .user_data      = NULL
    };
    vpi_register_systf(tf_data_ptr);

    // Register $get_matrix_a
    *tf_data_ptr = (s_vpi_systf_data){
        .type           = vpiSysFunc,
        .sysfunctype    = vpiSizedFunc,
        .tfname         = "$get_matrix_a",
        .calltf         = get_matrix_a_calltf,
        .compiletf      = NULL,
        .sizetf         = sized_function_64,
        .user_data      = NULL
    };
    vpi_register_systf(tf_data_ptr);

    // Register $get_matrix_b
    *tf_data_ptr = (s_vpi_systf_data){
        .type           = vpiSysFunc,
        .sysfunctype    = vpiSizedFunc,
        .tfname         = "$get_matrix_b",
        .calltf         = get_matrix_b_calltf,
        .compiletf      = NULL,
        .sizetf         = sized_function_64,
        .user_data      = NULL
    };
    vpi_register_systf(tf_data_ptr);

    free(tf_data_ptr); // Release the dynamically allocated memory
}
```

which sized_funcition_64 is
```C
PLI_INT32 sized_function_64(PLI_BYTE8 *user_data) {
    return 64; 
}
```

and I use array of `s_vpi_vecval` in callback function
```C
s_vpi_value val;
p_vpi_vecval result = (p_vpi_vecval)malloc(sizeof(s_vpi_vecval)*2);
 //some operation for result
val.format = vpiVectorVal;  // Setting format to vector since it's a 64-bit value
val.value.vector = result;  // Assuming suitable conversion here
vpi_put_value(systfref, &val, NULL, vpiNoDelay);
    
free(result);
```

But it only return 32 bit, not 64 bit.

I also tried to add `.sft` file while it reminded me `SFT files are deprecated. Please pass the VPI module instead.`

caryr, 2023-09-04:
The $realtobits() routine we provide is implemented using the VPI and returns a 64 bit value. You may want to try debugging this in pieces to narrow down where the actual issue is (e.g. send a known 64 bit value, etc.).

caryr, 2023-10-10:
Were you ever able to get this working? Can we close this bug report?

Devil-SX, 2023-10-11:
Sorry for the delayed response. Are you suggesting declaring it as a real-valued function in the C program and then calling $realtobits to convert it to 64 bits in Verilog?

caryr, 2023-10-11:
No I'm suggesting looking at the code in the `$realtobits()` implementation which should return a 64 bit value.

The building up process could be making a simple function that returns 64 bits of 0, then 64 bits of 1 and then maybe verify the individual words using alternate values. My guess is the size function is working correctly and it's how you are declaring and setting up the return value that is causing you issues. You could also try setting X and Z values to make sure you understand how to fill the result to get all possible 4-state values. If you show more of how the result is assigned that could help me figure out what is going wrong.

Also, is it just the value returned that makes you think your declaration is only returning 32 bits or is there a specific message that indicates the return value is the wrong size?

In general everything looks correct that you have shown so far, so I need more context to figure this out.

Devil-SX, 2023-10-11:
Thank you.
Here's how the result is assigned.

`$get_matrix_a`

```cpp
// s_vpi_vecval result[2]; // for 64-bit value
    p_vpi_vecval result = (p_vpi_vecval)malloc(sizeof(s_vpi_vecval)*2);
    result[0].aval = 0;
    result[0].bval = 0;
    result[1].aval = 0;
    result[1].bval = 0;
    
    for(int i=0; i<4; i++){
        int temp = - step + 3 + i;
        if(temp >= 0 && temp <= 3){
            if (i<2) {
                result[0].aval  |= (matrix_b[i][temp] << (i*16));
            } else {
                result[1].aval  |= (matrix_b[i][temp] << ((i-2)*16));    
            }
        }
    }
    vpi_printf("b:result[0].aval = %08x\n", result[0].aval);
    vpi_printf("b:result[1].aval = %08x\n", result[1].aval);

    val.format = vpiVectorVal;  // Setting format to vector since it's a 64-bit value
    val.value.vector = result;  // Assuming suitable conversion here
    vpi_put_value(systfref, &val, NULL, vpiNoDelay);

    free(result);
```

and my Verilog Testbench:

```verilog
  for (integer step= 0; step<2*4; step=step+1)
  begin
    $display("step = %d",step);
    @(posedge clk);
    a = $get_matrix_a(step);
    $display("a = %x",a);
    b = $get_matrix_b(step);
    print_matrix_32(result);
  end
```

and the output is ...

```
step =           2
a:result[0].aval = 000a0005
a:result[1].aval = 0000000f
a = 00000000000a0005
b:result[0].aval = 00070002
b:result[1].aval = 0000000c
[        52          0          0          0 ]
[         0          0          0          0 ]
[         0          0          0          0 ]
[         0          0          0          0 ]
```

`result[1].aval` was assigned correctly while `a` didn't.

And I've check the `$realtobits` implementation, arguments `type` , `sysfunctype` , `sizetf` are the same. And I don't know how `tf_data.compiletf` works

```
tf_data.type        = vpiSysFunc;
tf_data.sysfunctype = vpiSizedFunc;
tf_data.user_data   = "$realtobits";
tf_data.tfname      = tf_data.user_data;
tf_data.sizetf      = sizetf_64;
tf_data.compiletf   = sys_convert_compiletf;
tf_data.calltf      = sys_realtobits_calltf;
res = vpi_register_systf(&tf_data);
vpip_make_systf_system_defined(res);
```

caryr, 2023-10-11:
That all looks correct.

The compiletf routine is used to verify the argument types, number of argument, etc. since the compiler does not know what arguments a system task/function requires.

I'm really busy with work right now so it could be some time before I can build an example and test things locally. Though one other thing you could do is look at the output from the compiler and look for how `$get_matrix_a()` is called/declared. The compiler should have declared it 64 bits. This will help determine if it is a problem in the compiler defining the width of the function or in the runtime extracting the 64 bit value.

martinwhitaker, 2023-10-12:
Here is a complete example that works:

**test-vpi.c**
````
#include  "vpi_user.h"

static PLI_INT32 sizetf_64(PLI_BYTE8 *name)
{
    (void)name;  /* Parameter is not used. */
    return 64;
}

static PLI_INT32 copy64_compiletf(PLI_BYTE8 *name)
{
    vpiHandle call_handle = vpi_handle(vpiSysTfCall, 0);

    vpiHandle argv = vpi_iterate(vpiArgument, call_handle);
    vpiHandle arg;

    (void)name;  /* Parameter is not used. */

    /* Check that there is an argument. */
    if (argv == 0) {
	vpi_printf("copy64 requires one argument.\n");
        vpi_control(vpiFinish, 1);
	return 0;
    }

    arg = vpi_scan(argv);

    /* Check the argument is 64 bits. */
    if (vpi_get(vpiSize, arg) != 64) {
	vpi_printf("copy64 requires a 64-bit argument.\n");
        vpi_control(vpiFinish, 1);
	return 0;
    }

    /* Check there are no more arguments. */
    if (vpi_scan(argv)) {
	vpi_printf("copy64 only takes one argument.\n");
        vpi_control(vpiFinish, 1);
	return 0;
    }

    /* Save the argument away to make the calltf faster. */
    vpi_put_userdata(call_handle, (void *)arg);

    return 0;
}

static PLI_INT32 copy64_calltf(ICARUS_VPI_CONST PLI_BYTE8*name)
{
    vpiHandle call_handle = vpi_handle(vpiSysTfCall, 0);
    vpiHandle arg  = (vpiHandle)vpi_get_userdata(call_handle);
    s_vpi_value value;

    (void)name;  /* Parameter is not used. */

    /* Get the argument value */
    value.format = vpiVectorVal;
    vpi_get_value(arg, &value);

    vpi_printf("Copying aval %08x %08x bval %08x %08x\n",
      value.value.vector[1].aval,
      value.value.vector[0].aval,
      value.value.vector[1].bval,
      value.value.vector[0].bval
    );

    /* Return the same value. */
    vpi_put_value(call_handle, &value, 0, vpiNoDelay);

    return 0;
}

static void copy64_register(void)
{
      s_vpi_systf_data tf_data;

      tf_data.type        = vpiSysFunc;
      tf_data.sysfunctype = vpiSizedFunc;
      tf_data.user_data   = "$copy64";
      tf_data.tfname      = tf_data.user_data;
      tf_data.sizetf      = sizetf_64;
      tf_data.compiletf   = copy64_compiletf;
      tf_data.calltf      = copy64_calltf;

      (void)vpi_register_systf(&tf_data);
}

void (*vlog_startup_routines[])(void) =
{
    copy64_register,
    0
};
````
**test-vpi.v**
````
module test();

reg [63:0] value;

initial begin
  value = $copy64(64'h0123456789abcdef);
  $display("%h", value);
end

endmodule
````
Compiled and run as follows:
````
% iverilog-vpi test-vpi.c
Compiling test-vpi.c...
Making test-vpi.vpi from  test-vpi.o...
% iverilog -L . -m test-vpi -o test-vpi.vvp test-vpi.v 
% vvp test-vpi.vvp
Copying aval 01234567 89abcdef bval 00000000 00000000
0123456789abcdef
````
Note that you must supply the compiled VPI module to the compiler (`iverilog`). If you only supply it to the runtime (`vvp`), the compiler will not know the return type of the  user-defined system function and will assume it returns a 32-bit vector.

caryr, 2023-10-13:
I'm going to assume that you were not passing the file to the compiler and were getting the default width, which I had forgotten about. Please let us know if what Martin provided gives you enough information to solve your problem.

Devil-SX, 2023-10-14:
Martin's answer works.

My previous compile commands which getting 32-bit width: 

```shell
iverilog-vpi "vpi/matrix.c"
iverilog -g2012 -o sim_vpi.vvp -s [my_top_module] -y [my_dir] [some_verilog_files]
vvp -M "vpi" -m matrix  sim_vpi.vvp
```

And revised correct commands which getting expected 64-bit width:

```shell
iverilog-vpi "vpi/matrix.c"
iverilog -g2012 -o sim_vpi.vvp -s [my_top_module] -y [my_dir] -m "vpi/matrix.vpi" [some_verilog_files]
vvp sim_vpi.vvp
```

Why?

> Here is a complete example that works:
> 
> **test-vpi.c**
> 
> ```
> #include  "vpi_user.h"
> 
> static PLI_INT32 sizetf_64(PLI_BYTE8 *name)
> {
>     (void)name;  // Avoid warning about unused parameter.
>     return 64;
> }
> 
> static PLI_INT32 copy64_compiletf(PLI_BYTE8 *name)
> {
>     vpiHandle call_handle = vpi_handle(vpiSysTfCall, 0);
> 
>     vpiHandle argv = vpi_iterate(vpiArgument, call_handle);
>     vpiHandle arg;
> 
>     (void)name;  // Avoid warning about unused parameter.
> 
>     // Check that we have at least one argument.
>     if (argv == 0) {
> 	vpi_printf("copy64 requires one argument.\n");
>         vpi_control(vpiFinish, 1);
> 	return 0;
>     }
> 
>     arg = vpi_scan(argv);
> 
>     // Check the argument is a 64 bit vector.
>     if (vpi_get(vpiSize, arg) != 64) {
> 	vpi_printf("copy64 requires a 64-bit argument.\n");
>         vpi_control(vpiFinish, 1);
> 	return 0;
>     }
> 
>     // Check there are no more arguments.
>     if (vpi_scan(argv)) {
> 	vpi_printf("copy64 only takes one argument.\n");
>         vpi_control(vpiFinish, 1);
> 	return 0;
>     }
> 
>     // Save the argument handle for later use.
>     vpi_put_userdata(call_handle, (void *)arg);
> 
>     return 0;
> }
> 
> static PLI_INT32 copy64_calltf(ICARUS_VPI_CONST PLI_BYTE8*name)
> {
>     vpiHandle call_handle = vpi_handle(vpiSysTfCall, 0);
> 
>     vpiHandle arg = (vpiHandle)vpi_get_userdata(call_handle);
> 
>     s_vpi_value value;
> 
>     (void)name;  // Avoid warning about unused parameter.
> 
>     // Get the argument value.
>     value.format = vpiVectorVal;
>     vpi_get_value(arg, &value);
> 
>     vpi_printf("Copying aval %08x %08x bval %08x %08x\n",
>       value.value.vector[1].aval,
>       value.value.vector[0].aval,
>       value.value.vector[1].bval,
>       value.value.vector[0].bval
>     );
> 
>     // Return the same value.
>     vpi_put_value(call_handle, &value, 0, vpiNoDelay);
> 
>     return 0;
> }
> 
> static void copy64_register(void)
> {
>       s_vpi_systf_data tf_data;
> 
>       tf_data.type        = vpiSysFunc;
>       tf_data.sysfunctype = vpiSizedFunc;
>       tf_data.user_data   = "$copy64";
>       tf_data.tfname      = tf_data.user_data;
>       tf_data.sizetf      = sizetf_64;
>       tf_data.compiletf   = copy64_compiletf;
>       tf_data.calltf      = copy64_calltf;
> 
>       (void)vpi_register_systf(&tf_data);
> }
> 
> void (*vlog_startup_routines[])(void) =
> {
>     copy64_register,
>     0
> };
> ```
> 
> **test-vpi.v**
> 
> ```
> module test();
> 
> reg [63:0] value;
> 
> initial begin
>   value = $copy64(64'h0123456789abcdef);
>   $display("%h", value);
> end
> 
> endmodule
> ```
> 
> Compiled and run as follows:
> 
> ```
> % iverilog-vpi test-vpi.c
> Compiling test-vpi.c...
> Making test-vpi.vpi from  test-vpi.o...
> % iverilog -m ./test-vpi.vpi -o test-vpi.vvp test-vpi.v 
> % vvp test-vpi.vvp
> Copying aval 01234567 89abcdef bval 00000000 00000000
> 0123456789abcdef
> ```
> 
> Note that you must supply the compiled VPI module to the compiler (`iverilog`). If you only supply it to the runtime (`vvp`), the compiler will not know the return type of the user-defined system function and will assume it returns a 32-bit vector.

martinwhitaker, 2023-10-14:
> Martin's answer works.
> 
> My previous compile commands which getting 32-bit width:
> 
> ```shell
> iverilog-vpi "vpi/matrix.c"
> iverilog -g2012 -o sim_vpi.vvp -s [my_top_module] -y [my_dir] [some_verilog_files]
> vvp -M "vpi" -m matrix  sim_vpi.vvp
> ```
> 
> And revised commands which getting expected 64-bit width:
> 
> ```shell
> iverilog-vpi "vpi/matrix.c"
> iverilog -g2012 -o sim_vpi.vvp -s [my_top_module] -y [my_dir] -m "vpi/matrix.vpi" [some_verilog_files]
> vvp sim_vpi.vvp
> ```
> 
> Why?

See https://steveicarus.github.io/iverilog/usage/vpi.html#system-function-return-types

closed 2023-10-14 by Devil-SX