steveicarus/iverilog #991: Odd behaviour signl behaviour inside a generate loop.

sean-galloway, 2023-08-19:
Hi, I have a full cocotb directory for the passing and failing situations if needed. Here is a synopsis of the issue. In the failing case, I created a "dummy" carry_in signal to override a value. I ditched the dummy signal in the passing case and attached the muxes straight to the port. What happens in the failing case is, the carry_in signal return Z even though all of the components of the signal are know. This is proven in the passing case where the muxes are directly connected to the port of the CLA instance.

**Failing code:**
// Stages of addition
   generate
       for(stage = 0; stage < STAGES; stage=stage+1) begin
           for(i = 0; i < C_PADDED >> stage; i=i+2) begin
               **logic carry_in = (stage == 0) ? 1'b0: (i == 0) ? 1'b0 : w_c[stage][i];**      
               math_adder_carry_lookahead #(N) adder_inst (
                   .i_a(w_intermediate_sums[stage][i]),
                   .i_b(w_intermediate_sums[stage][i + 1]),
                   .i_c(**carry_in**),
                   .ow_sum(w_intermediate_sums[stage + 1][i/2]),
                   .ow_c(w_c[stage+1][i/2])
               );
           end
       end
   endgenerate

**Passing code:**
   // Stages of addition
   generate
       for(stage = 0; stage < STAGES; stage=stage+1) begin
           for(i = 0; i < C_PADDED >> stage; i=i+2) begin
               math_adder_carry_lookahead #(N) adder_inst (
                   .i_a(w_intermediate_sums[stage][i]),
                   .i_b(w_intermediate_sums[stage][i + 1]),
                   .i_c(**(stage == 0) ? 1'b0: (i == 0) ? 1'b0 : w_c[stage][i]**),
                   .ow_sum(w_intermediate_sums[stage + 1][i/2]),
                   .ow_c(w_c[stage+1][i/2])
               );
           end
       end
   endgenerate
[IcarusIssue.tar.gz](https://github.com/steveicarus/iverilog/files/12385460/IcarusIssue.tar.gz)

larsclausen, 2023-08-19:
The two examples are not equivalent.

In the first example it is an initializer expression that gets evaluated once. The other one is a continuous assignment that gets evaluated whenever the expression changes.

If you change the first example to
```SystemVerilog
logic carry_in;
assign carry_in = (stage == 0) ? 1'b0: (i == 0) ? 1'b0 : w_c[stage][i];
```
you should get the same result as for the second example.

sean-galloway, 2023-08-19:
Hi,
I thought I tried that, but I’ll check it out. Thanks for the quick response!

-sean

caryr, 2023-08-23:
Did adding the `assign` solve your problem?

caryr, 2023-09-04:
No more information provided so closing this report.

closed 2023-09-04 by caryr