steveicarus/iverilog #1001: Unpacked array output port result in undefined

smunaut, 2023-09-21:
I'm trying to use an output port that's an unpacked array and all the output values are `xxxx`.
I'm using the following test bench compiled with `iverilog test.sv -o test -g2012`. 

```systemverilog
`default_nettype none
`timescale 1 ns / 100 ps

module test_tb;

	// Signals
	logic rst = 1;
	logic clk = 0;

	logic [15:0] out_data [0:3];

	// Setup recording
	initial begin
		$dumpfile("test_tb.vcd");
		$dumpvars(0,test_tb);
	end

	// Reset pulse
	initial begin
		# 201 rst = 0;
		# 65000 $finish;
	end

	// Clock
	always #5 clk = !clk;

	// DUT
	test dut_I (
		.out_data(out_data),
		.clk(clk),
		.rst(rst)
	);

	// Show result
	always @(posedge clk)
		if (~rst)
			$display("%x", out_data[0]);

endmodule // test_tb

module test (
	output logic [15:0] out_data [0:3],
	input wire clk,
	input wire rst
);

	logic [15:0] cnt;

	assign out_data[0] = cnt;
	assign out_data[1] = cnt + 1;
	assign out_data[2] = cnt + 2;
	assign out_data[3] = cnt + 3;

	always @(posedge clk)
		if (rst)
			cnt <= 0;
		else
			cnt <= cnt + 1;

endmodule // test
```

smunaut, 2023-09-21:
Just saw that if I use `wire [15:0] out_data [0:3];` instead of `logic [15:0] out_data [0:3];`  it works ...

smunaut, 2023-11-07:
Anyone got a clue about this one ?

caryr, 2023-11-07:
I don't think anyone has looked at this yet. This is a more complicated issue where we need to look at your code and the code generated for vvp to see what is going wrong. I and I assume no one else has had time to do this yet.

smunaut, 2023-11-15:
The relevant difference in the produced vvp script seems to be :

```diff
-v0x55d1d9816bf0 .array "out_data", 3 0;
-v0x55d1d9816bf0_0 .net v0x55d1d9816bf0 0, 15 0, L_0x55d1d9826f90; 1 drivers
-v0x55d1d9816bf0_1 .net v0x55d1d9816bf0 1, 15 0, L_0x55d1d9827310; 1 drivers
-v0x55d1d9816bf0_2 .net v0x55d1d9816bf0 2, 15 0, L_0x55d1d98273d0; 1 drivers
-v0x55d1d9816bf0_3 .net v0x55d1d9816bf0 3, 15 0, L_0x55d1d9827490; 1 drivers
+v0x557f5c2b9bc0 .array "out_data", 3 0, 15 0;
```

I also notice when enabling elaborate debug that the net type for `out_data` becomes `REG` in the non-working case.

smunaut, 2023-11-16:
This seems to fix the issues and doesn't break the tests but I have no clue if that's even remotely the right thing to do ...

```diff
diff --git a/elaborate.cc b/elaborate.cc
index 1467fa7f1..1ad487477 100644
--- a/elaborate.cc
+++ b/elaborate.cc
@@ -1206,9 +1206,10 @@ void elaborate_unpacked_port(Design *des, NetScope *scope, NetNet *port_net,
       }
 
       ivl_assert(*port_net, expr_net->pin_count() == port_net->pin_count());
-      if (port_type == NetNet::POUTPUT)
+      if (port_type == NetNet::POUTPUT) {
+           expr_net->type(NetNet::WIRE);
            assign_unpacked_with_bufz(des, scope, port_net, expr_net, port_net);
-      else
+      } else
            assign_unpacked_with_bufz(des, scope, port_net, port_net, expr_net);
 }
 
```

martinwhitaker, 2024-01-30:
Your fix was on the right lines. We need to convert to an `UNRESOLVED_WIRE`, not a WIRE`, to catch multiple drivers. And we must take care not to change the net type if the destination is already declared as a net.

Fixed in the master branch.

closed 2024-01-30 by martinwhitaker