steveicarus/iverilog #1004: Lack of parameter expansion causing warnings

sidmontu, 2023-09-28:
Take a minimal example below:

```sv
module minimal ();
    localparam FOO = 3;

    logic v [0:FOO-1];
    logic nv [0:FOO-1];

    integer i; 
    always_comb begin
        for (i=0; i<FOO; i=i+1) begin
            nv[i] = 1'b0;
            if (v[i]) nv[i] = 1'b1;
        end
    end

endmodule: minimal
```
`iverilog` compiles above successfully without any warnings or errors, as expected.

Now if I change the `localparam FOO = 3` declaration to `localparam FOO = 3 + 1`, I get the following warning:

```
minimal.sv:9: warning: A for statement must use the index (i) in the condition expression to be synthesized in an always_comb process.
```

Observation: Any form of parameter expansion prints the above warning. Examples: `parameter FOO = BAR + BAZ`, `localparam FOO = BAR + 1`, etc.

Tested with: `iverilog v12.0 (devel) (s20150603-1658-g74c52d6fa)` and  `iverilog v13.0 (devel) (s20221226-114-ge9646bbbd)`. Command used: `iverilog -g2012 -ominimal minimal.sv`

larsclausen, 2023-10-29:
Thanks for the report. This is a duplicate of #687. The issue occurs if the width of the index is less than the width of the value it is being compared to.

martinwhitaker, 2024-01-07:
Fixed in the master branch. If you are using a stable release, you can suppress this warning in this particular case by using the `-gstrict-expr-width` compiler option, which stops the constant expression being expanded beyond 32 bits.

closed 2024-01-07 by martinwhitaker