steveicarus/iverilog #1009: [Request] Add built executables to the GitHub releases (esp. for Windows)

DeflateAwning, 2023-10-16:


caryr, 2025-07-18:
The other issues have more details, but this one was first. Specifically #1250 points to work work Nick is doing on nvc that we may be able to leverage to automatically generate binaries.

srcejon, 2026-10-29:
> I have been investigating how to sign binaries and that is too complicated, so these will likely be unsigned and people will have to deal if Windows will accept them.

[SignPath.io](https://signpath.org/) will sign open-source apps for free - and can be integrated in to a github build action (I use it for another app).

caryr, 2026-10-29:
Thanks for the reference! I will look into this. I guess getting an action that can trigger building and signing the installer is a high priority. Does anyone have a reference that can be followed for this action?