steveicarus/iverilog #1012: vvp: internal error when concatenating verilog array elements

proppy, 2023-10-20:
The following verilog:
```
module ArraySliceWithNarrowStart(
  input wire [159:0] a,
  input wire start,
  output wire [95:0] out
);
  wire [31:0] a_unflattened[0:4];
  assign a_unflattened[0] = a[31:0];
  assign a_unflattened[1] = a[63:32];
  assign a_unflattened[2] = a[95:64];
  assign a_unflattened[3] = a[127:96];
  assign a_unflattened[4] = a[159:128];
  wire [31:0] array_slice_6[0:2];
  assign array_slice_6[0] = a_unflattened[{2'h0, start} > 3'h4 ? 3'h4 : {2'h0, start} + 3'h0];
  assign array_slice_6[1] = a_unflattened[{2'h0, start} > 3'h3 ? 3'h4 : {2'h0, start} + 3'h1];
  assign array_slice_6[2] = a_unflattened[{2'h0, start} > 3'h2 ? 3'h4 : {2'h0, start} + 3'h2];
  assign out = {array_slice_6[2], array_slice_6[1], array_slice_6[0]};
endmodule
```
trigger the following error in vvp:
```
internal error: port 0 expects wid=32, got wid=0
vvp: concat.cc:54: virtual void vvp_fun_concat::recv_vec4(vvp_net_ptr_t, const vvp_vector4_t&, vvp_context_t): Assertion `0' failed.
```

proppy, 2023-10-20:
managed to reduce the test case to the following:
```
module ArraySliceConcat(
  input wire start,  		      
  output wire [32:0] out
);
  wire [31:0] a_unflattened[0:4];
  assign out = a_unflattened[{2'h0, start} > 3'h4 ? 3'h0 : 3'h0];
endmodule
```

proppy, 2023-10-20:
seems to be related to the conditional indexing evaluation: `a_unflattened[{2'h0, start} > 3'h4 ? 3'h0 : 3'h0]`.

proppy, 2023-10-20:
fyi, it doesn't seem to fail with https://github.com/steveicarus/iverilog/tree/v11_0

proppy, 2023-10-20:
nor with with https://github.com/steveicarus/iverilog/tree/v12_0

larsclausen, 2023-10-20:
https://github.com/steveicarus/iverilog/commit/a7883aaf7d55b579514bda4509f5103a28402665 should fix it. Will create a PR later.

proppy, 2023-10-20:
@larsclausen do you want me to submit the snippet in this bug as an additional test?

larsclausen, 2023-10-21:
> @larsclausen do you want me to submit the snippet in this bug as an additional test?

That sounds great, let me know when that is ready and I can create PR with both the fix and the test so CI can test them both together.

Having had some more time to think about the problem I went with a different fix https://github.com/larsclausen/iverilog/commit/d0f3c4db6458a6a8861206952995785e77103543

proppy, 2023-10-23:
> That sounds great, let me know when that is ready

Done! https://github.com/steveicarus/iverilog/pull/1014

caryr, 2023-11-07:
@larsclausen can you verify if the test code is working with your latest fix and if so get your change along with the test into the repo?

larsclausen, 2024-01-09:
Marking this as fixed since both the fix and the testcase have been merged.

closed 2024-01-09 by larsclausen