steveicarus/iverilog #1015: explicit cast should not be required

jotego, 2023-10-27:
In the following code:

```
if( is_im_logic ) md_ctrl = fetch_next_md; else md_ctrl = fetch_first_md;
md_ctrl    = (is_im_logic ? fetch_next_md : fetch_first_md);
```
where md_ctrl is an enumerated type, the first line will be parsed correctly while the second one produces the error `This assignment requires an explicit cast.`

I think both lines are equivalent and the error should not occur. 
This is tested in version Icarus Verilog version 12.0 (stable) (s20221226-36-g23e51ef7a)

larsclausen, 2023-10-29:
Its a bit tricky. The two are not equivalent, there is a difference in behavior when then condition has a `x` value. (System)Verilog requires that the two operands for a ternary operator get combined bit by bit when the condition is undefined. For an enum this can result in a value that is not part of the enumeration.

E.g. in the following example `c1` and `c2` will have different values.
```SystemVerilog
module test;

  logic x;
  typedef enum {
    A = 1,
    B = 2
  } E;

  E a, b, c1, c2;

  initial begin
    a = A;
    b = B;
    x = 1'bx;

    c1 = E'(x ? a : b);

    if (x) c2 = a; else c2 = b;

    $display(c1, c2);
  end

endmodule
```

But maybe this should just be a warning instead of an error.

jotego, 2023-10-30:
That is very interesting. These specs have some quirks unknown to many of us! 

Thank you for the explanation.

closed 2023-10-30 by jotego