steveicarus/iverilog #1016: break in for loop doesn't seem to execute

edanuff, 2023-10-29:
Working with the following function:

```
  function [PORT_BITS:0] get_priority_port();
    bit result = 0;
    for (int i = 0; i < NUM_PORTS; i++) begin : priority_loop
      $display("Port %d is %b", i, port_req[i] || port_req_queue[i]);
      if (port_req[i] || port_req_queue[i]) begin
        result = i;
        break;
      end
    end
    $display("Priority port identified as %b", result);
    return result;
  endfunction
```

I always see the following output, where the loop is executing for each item:

```
Port           0 is 1
Port           1 is 1
Priority port identified as 1
```

I have built from master to make sure I'm running the latest version.  I assume master has #855 incorporated?

larsclausen, 2023-10-29:
There is unfortunately still an issue where break/continue do not work within named begin/end blocks or within blocks that have variables declared inside them.

As a workaround you can remove the name from your for loop.

```
  function [PORT_BITS:0] get_priority_port();
    bit result = 0;
    for (int i = 0; i < NUM_PORTS; i++) begin
      $display("Port %d is %b", i, port_req[i] || port_req_queue[i]);
      if (port_req[i] || port_req_queue[i]) begin
        result = i;
        break;
      end
    end
    $display("Priority port identified as %b", result);
    return result;
  endfunction
  ```

edanuff, 2023-10-29:
Ah, that fixed it, thanks!

larsclausen, 2023-10-29:
Lets keep the issue open, since this an actual bug in iverilog that needs to be fixed.

kurochan001, 2026-10-10:
On current master the same bug can also make a `while` loop run away instead of only ignoring the `break`. When the `break` is inside a nested `begin`-`end` block that declares a variable (or is named), the statements after the loop are executed, and then the loop keeps running:

```systemverilog
module top;
  int n;
  initial begin
    n = 0;
    while (1) begin
      n++;
      if (n > 5) begin
        $display("FAILED: loop still running, n=%0d", n);
        $finish;
      end
      begin
        int k;          // a declaration in the block that contains the break
        break;
      end
    end
    $display("after loop: n=%0d (expected 1)", n);
  end
endmodule
```

```
$ iverilog -g2012 -o test.vvp test.sv && vvp test.vvp
after loop: n=1 (expected 1)
after loop: n=2 (expected 1)
after loop: n=3 (expected 1)
after loop: n=4 (expected 1)
after loop: n=5 (expected 1)
FAILED: loop still running, n=6
test.sv:9: $finish called at 0 (1s)
```

Without the `if (n > 5)` guard the loop never ends and prints `after loop:` lines without limit. Removing the declaration `int k;` gives the expected single `after loop: n=1`; naming the inner block (`begin : blk`) instead of declaring `k` gives the same runaway output as above.

With a `for` loop of the same shape, the `break` is ignored as described in this issue (the loop runs all 4 iterations and `after loop: cnt=4 (expected 1)` is printed once).

IEEE 1800-2023 12.8: "The break statement jumps out of the loop."

Version: `Icarus Verilog version 14.0 (devel) (s20260301-541-gb25510638)`, current master built from source, on macOS (Apple Silicon).

---

This report was prepared with Claude Code (Anthropic's AI coding assistant) using the model Claude Opus 5.5 (`claude-opus-5-5`). The reproductions were reduced and run with the build listed above, and the outputs are copied from those runs.