steveicarus/iverilog #1019: [Request] Add `--help` and `--version` args to `iverilog`

DeflateAwning, 2023-11-03:
Add `--help` and `--version` args to `iverilog`

These are very common test commands, which would be very helpful to have setup

martinwhitaker, 2023-11-05:
We already have `-h` and `-V`.

DeflateAwning, 2023-11-06:
Should be easy to extend then. The trend seems to be supporting these long-form ones, and even moving away from the shorter versions as those flags are non guaranteed to do other things.

I may work on implementing this!

caryr, 2023-11-06:
Having the long-forms would  be nice, though we need to make sure this works on all systems. In the past this was handled by adding the appropriate routines from the GNU source so that they can be used if the system does not fully support the long-form. If you want to take this on look at what is implemented for the short-form. Adding the long form would be helpful for a few other tickets.

DeflateAwning, 2023-11-06:
Hmm quirky that it may cause issues? Can you point to the part of the code that handles CLI args (still getting familiar with it)? I wonder if we should start using a library to handle CLI args; it's very common in other languages/projects. Any thoughts?

martinwhitaker, 2023-11-06:
> **Edit:** Found [the flag handler part](https://github.com/steveicarus/iverilog/blob/77d7f0b8f73de3111580294744e3e95c19a0aa30/main.cc#L966). Will investigate.

That's for the main compiler, `ivl`, which is not normally executed directly. The`iverilog` command is a wrapper that first calls the pre-processor, then the main compiler. Look in the `driver` subdirectory, specifically `driver/main.c`.