steveicarus/iverilog #1021: [Request] Add Windows builds to releases

DeflateAwning, 2023-11-03:


caryr, 2023-11-29:
For the most part we have not provided binaries, but I certainly understand this would be helpful. We are all very busy with our day jobs so have little time for Icarus lately and adding build management and supporting questions related to the binaries is likely something we do not have time for. I do agree this would be beneficial if we had the bandwidth to support it.

If you want to do the research into how github could automate this that would be awesome!

Kreijstal, 2024-06-16:
@DeflateAwning you can just use msys2 version and package manager

DeflateAwning, 2024-06-17:
I understand that there are ways to get builds on Windows; I'm just suggesting that there's a standard and accepted way to manage releases with GitHub, and attaching Windows build products is the accepted way. Would love to see this project support the de-factor release mechanism for Windows.

caryr, 2025-07-18:
Closing this as a duplicate of #1009