steveicarus/iverilog #1022: Port array assignment support to 12.0 branch

Bob0111, 2023-11-04:
According to the issue #562, the array assignment was added to the iverilog not long before. However, the code was only merged into the master branch instead of the actual release branch and since many Linux distribution relies on the release branch, they can't get benefit from that. Is it possible to port the function to the old release branch?It would be good to support that in the release branch as it is a standard in the systemverilog.

larsclausen, 2023-11-06:
I understand your point, but it is unfortunately not that easy.

Many of the features being added build on each other and a series of refactorings. Taking a single PR and adding it to a release is often not possible and also requires adding everything that lead up to it.

Also which features are important enough to be added to the release branch? Most new features work towards SystemVerilog language completion.

The goal is to try to have releases on a regular cadence to make sure new features propagate to distributions in a reasonable timeframe.

caryr, 2023-11-06:
We are planning to have a release by the end of the year. Until then compiling the latest code from the development branch is always the best way to have access to the latest changes.

While we do not guarantee stability in the development branch, because of our extensive testsuite, development is almost always stable and if it is unstable it is usually only for a very short period of time.

Bob0111, 2023-11-09:
OK. So I'd also want to participate in the project to make the new release happen earlier, but I'm new to the project. How can I get started?

martinwhitaker, 2024-01-08:
Closing as not planned. v13 should be released soon.

closed 2024-01-08 by martinwhitaker