steveicarus/iverilog #1026: Install error for v12.0

arunpatro, 2023-11-13:
When I checkout the v12-branch and run 

```sh
sh autoconf.sh
./configure
make
```
I get 
```sh
make: *** No rule to make target 'PTimingCheck.h', needed by 'main.o'.  Stop.
```

How can I resolve this? 

The main branch works as expected and I get v13.0 installed.

caryr, 2023-11-13:
I'm going to guess you are building the V12 branch after building the V13/master branch in the same build directory and there are leftover dependencies for the V13/master build leaking into the V12 build. I always build these in separate directories to keep things separated, but if you do a make distclean before switching to the V12 branch that should remove the dependency files.

If this is not the case then we likely need more information regarding the exact steps you are doing.

arunpatro, 2023-11-13:
Aah I see my mistake. `make distclean` works :)

closed 2023-11-13 by arunpatro