steveicarus/iverilog #1028: Icarus verilog makes no progress

rowanG077, 2023-11-14:
I have this `tb.v`:

```verilog
`timescale 1ns / 1ps

module tb;

  initial begin
    #100;
    $finish;
  end

  colorlite inst ();

endmodule
```

And this `colorlite.v`:

```verilog
`timescale 1ns / 1ps
module colorlite ( );

reg           packetizer_is_ongoing2 = 1'd0;
reg           packetizer_sink_ready = 1'd0;
wire          packetizer_sink_valid = 1'd0;
reg           packetizer_source_last_a = 1'd0;
reg           packetizer_source_last_b = 1'd0;

always @(*) begin
    packetizer_is_ongoing2 <= 1'd0;
    if (packetizer_sink_valid) begin
        packetizer_source_last_a <= 1'd0;
    end
    packetizer_is_ongoing2 <= 1'd1;

    if (packetizer_is_ongoing2) begin
        packetizer_source_last_b <= 1'b0;
    end
end

endmodule
```

Which is a severely cut down version of the verilog generated by litex from https://github.com/rowanG077/liteeth-16-byte-bug/tree/testing

I run the following commands:

```
iverilog -o sim.vvp -s tb -Wall -pfileline=1 colorlite.v tb.sv
vvp sim.vvp
```

(Apparently `-pfileline=1` should add lot's of debug output but It doesn't show anything).

I'm running icarus verilog 12.0 on NixOS aarch64

closed 2023-11-14 by rowanG077

caryr, 2023-11-14:
What seems odd to me is the non-blocking assignments to `packetizer_is_ongoing2` should those be blocking assignments? That looks like the basis for the infinite looping.

rowanG077, 2023-11-14:
Considering it is generated by litex they should be correct. It's also weird that removing the first if makes it work as well.