steveicarus/iverilog #1030: Arithmetic shift changes to logic shift after wait statement

Byter64, 2023-11-14:
When using an if statement on a wire (here "result") the arithmetic right shift operator does not return an arithmetic shift anymore but a logic shift after a wait statement. Even if the wait statement is #0 (tested with #0, #1, #200);

This is the minimum required code, that is needed in order to reproduce this bug.
At the first if, the result is as expected  "**_1_** 1100100000011001010101010101010"
At the second if the result is unexpectedly "**_0_** 1100100000011001010101010101010" .

Code:
```verilog
module Bug;

localparam shiftRightArithmetic = 6'b000111;
localparam equals               = 6'b001010;

reg[5:0]  code = shiftRightArithmetic;
reg[31:0] bitstring = 32'b11001000000110010101010101010101;
reg[31:0] shiftAmount = 32'd1;

wire[31:0] result = code[5:0] == shiftRightArithmetic   ? $signed(bitstring) >>> $signed(shiftAmount) : 
                    code[5:0] == equals                 ? $signed(bitstring) == $signed(shiftAmount) : 0;

initial begin
    if(result != 32'b11100100000011001010101010101010) //result is fine
        $display("1 Arithmetic right-shift not working: \n Bitstring:%b \n places:  %d \n result:   %b \n expected: %b\n", bitstring, shiftAmount, result, 32'b11100100000011001010101010101010);
    #0 
    if(result != 32'b11100100000011001010101010101010) //result is not fine
        $display("2 Arithmetic right-shift not working: \n Bitstring:%b \n places:  %d \n result:   %b \n expected: %b\n", bitstring, shiftAmount, result, 32'b11100100000011001010101010101010);
end

endmodule
```

result from command line:
[image]

martinwhitaker, 2023-11-14:
There is a race in your code. If you change your test from `result !=` to `result !==`, you will see that `result` is `x` before the wait. That is because the initial block can be executed before the continuous assignment is evaluated.

The reason you aren't getting the result you expect is because `$signed(bitstring) == $signed(shiftAmount)` results in a single bit unsigned value. That makes the entire expression unsigned. Changing that part to `$signed(bitstring == shiftAmount)` will fix that.

Byter64, 2023-11-15:
Thanks a ton :)
Sorry for branding this directly as a bug (of icarus verilog). I didn't see the race condition there nor did I know that an unsigned value within an if statement makes that if statement's result unsigned. Thanks again