steveicarus/iverilog #1038: cannot access package objects with vpi_handle_by_name()

toddstrader, 2023-11-29:
I tried both a `::` and `.` delimiter (I believe the former to be the correct choice), but neither worked:
```
    vpiHandle h1 = vpi_handle_by_name("cocotb_package_pkg_1::five_int", NULL);
    fprintf(stderr, "h1 = %u\n", h1);
    vpiHandle h2 = vpi_handle_by_name("cocotb_package_pkg_1.five_int", NULL);
    fprintf(stderr, "h2 = %u\n", h2);
```
yields:
```
h1 = 0
h2 = 0
```

However, iterating over objects in the package does work.

Related to:
https://github.com/cocotb/cocotb/pull/3536

caryr, 2023-11-29:
The code for vpi_handle_by_name() likely predates when packages were added so it will need some tweaking to handle them correctly.

caryr, 2023-12-29:
I think I have a fix for this, but I want to review it again tomorrow before I turn it in. It also allows you to find packages by name. They need to have `::` as a prefix to make it obvious what you are looking for and I have verified you can find variables in a package by name which is what you originally reported (e.g. `vpi_handle_by_name("pkg::var", NULL);` should work in the development branch once this is turned in). I doubt this change will be back ported to 12, but I hope we will release 13 soon and it's very unlikely we will release12.1 given we would like to get 13.0 out.

Do you have any information regarding how other simulators work in this respect? For example does `vpi_handle_by_name("pkg::var", NULL);` return a handle to the variable named `var` in the package named `pkg` and does `vpi_handle_by_name("pkg::", NULL);` return a handle to the package named `pkg`? This seems like the most obvious implementation, but I do not see where it is actually specified in the standard to work this way.

toddstrader, 2023-12-29:
"pkg::var" works on xcelium, questa and vcs but makes riviera crash.  Questa and xcelium can find "pkg::" while riviera and vcs return NULL.

steveicarus, 2023-12-29:
Close enough. I say go with Cary's solution.

caryr, 2023-12-29:
Thanks for the comparison information. That confirms the implementation plan I'm following should be mostly portable.

I still need to check escaped package identifiers work and would like to refactor the code a bit because it's slightly hackish even though it works correctly. For example I believe there is only a single package base and I've not looked at optimizing that along with my detection method for a package needs some more robustness if possible. I think an escaped module identifier that ends in :: would be considered a package so would not be search for in the module namespace, but it may be protected by the trailing space until after type detection.

Just more test cases to add and confirm they work correctly. At the moment I am confirming you can access the scope by name and a single `int` variable in each of the domains.