steveicarus/iverilog #1046: Iverilog hang on very large array of flops

jordankrim, 2023-12-14:
I am running Iverilog and I have an extremely large multi array of flops generated via a genvar statement.  When I start my simulation, Iverilog hangs possibly attempting to do an asynchronous reset on this very large array.  I know this because when I comment this module out, the simulation does not hang and proceeds after the asynchronous  reset is deactivated.  Is there any way to (1) deug this, a,though since I know where it is, debug may not help much and (2) somehow coax Iverilog (via parms or something) to get past this hang?

martinwhitaker, 2023-12-15:
Compile your code with `-pfileline=1` then run the code with `vvp -s`. At the interactive prompt enter the commands
````
trace
cont
````
That will print out file and line numbers for each behavioural statement as it is executed. If the output stops, that will show the last statement executed. If the output continues indefinitely, use Ctrl-C to interrupt it, and you can then scroll back to see what statements it's looping over.

caryr, 2024-01-07:
Any update on getting us an example we can use to see if this is a bug? In general we need, preferably simplified, code to figure out what is causing the behavior you are experiencing.

jordankrim, 2024-01-08:
I had to abandon the RTL that I thought was causing this issue, not just because of the potential hang, but because it was too slow in simulation and very large in structure.  I found a better solution and am moving forward with that.

martinwhitaker, 2024-01-09:
OK, closing as we have no way to reproduce the issue.

closed 2024-01-09 by martinwhitaker