steveicarus/iverilog #1047: Iverilog not initializing the register with initial value

abdulhameed-rs, 2023-12-15:
The DFF has 0 initial value assign as shown below:
[image]

When simulation is run on iverilog vs VCS there is a difference of results shown in waveform below. The i1[0] is output of the flop which is triggered at negedge of clock, before that negedge of clock occurs the output should go zero (as initialized). This initial value can be seen in VCS simulation waveform below:

VCS waveform:
[image]

When iverilog simulation is run it does not get the initial value of 0, instead there is unknown value before first negedge. shown below in waveform below:

Iverilog waveform:
[image]

martinwhitaker, 2023-12-15:
I can't reproduce this. Please provide a simple test case that demonstrates this behaviour. Please provide your code directly as text (using markdown code tags), not as an embedded image, so we can copy and paste it.

What version of iverilog are you using?

abdulhameed-rs, 2023-12-19:
I am using iverilog version:
Icarus Verilog version 13.0 (devel) (s20221226-241-g999bcb6)

I was simulating a netlist which was using the module:
module DFFNRE (
  input D, 
  input R, 
  input E, 
  input C,
  output reg Q = 1'b0 
);

  always @(negedge C, negedge R)
    if (!R)
      Q <= 1'b0;
    else if (E)
      Q <= D;

endmodule

martinwhitaker, 2023-12-19:
As I said, I can't reproduce the problem. I used this test bench code
````
module test;

reg D, R, E, C;
wire Q;

DFFNRE dff(D, R, E, C, Q);

initial begin
  $monitor(D,,R,,E,,C,,Q);
  D = 0;
  E = 0;
  R = 1; 
  C = 1; 
  #1 $finish;
end

endmodule
````
which when run outputs this
````
0 1 0 1 0
test.v:30: $finish called at 1 (1s)
````
showing that Q has been initialised to 0 as expected.

If you provide a test case that demonstrates the fault, we can investigate further.

caryr, 2024-01-07:
Have you been able to see if the test code you are using is what is causing the issue? We ideally like a complete, preferably simplified, code example that demonstrates the problem. When we have to make guesses or assumptions about missing code debugging is harder or not possible.

caryr, 2025-07-15:
No response given so closing this.